bambulab/BambuStudio #4984: Jerk(XY) can be searched in Process settings but not editable

caizixian, 2024-10-13:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 15.0.1

### Printer

Bambu Lab A1

### How to reproduce

1. Click Process (Global)
2. Click the search icon
3. Type "jerk"
4. Click one of the search results, such as "Jerk (XY): outer wall"

### Actual results

It goes to the speed table, but there's no Jerk (XY) setting.

### Expected results

Able to edit Jerk (XY) settings.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

caizixian, 2024-10-13:
See also https://www.reddit.com/r/BambuLab/comments/19e9wyl/how_do_i_set_these_jerk_values/

caizixian, 2024-10-13:
Looks like it's disable for Bambu printers https://github.com/bambulab/BambuStudio/commit/aa62c6e870535eaa7b9b8b5a35aecfff93f6fb7a#diff-cb3d09d2073b6e9032c65d07aa8f10d166324a53136b2aa1288f7b76f7d94319L1877

bambu123, 2024-10-14:
I will check that.

caizixian, 2025-09-02:
I can still observe this in 2.2.1.60