bambulab/BambuStudio #4985: Channing world coordinates lead to completely diffenent values as entered

shootster, 2024-10-13:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.0.1

### Additional system information

Mac Book Pro 14" 2023, M2 Pro, 16 GB Ram

### Printer

BambuLab P1S

### How to reproduce

1. select sth. on the plate
2. select "changing world coordinates" in the toolbar
3. enter value for x/y/z and confirm by pressing enter

### Actual results

The changed value than is completely different. e.g. current value for Xis 1289.12. I enter 900 and after confirmation it is 1678.24

### Expected results

value for X should be 900

### Project file & Debug log uploads

[SETI-Insert.3mf.zip](https://github.com/user-attachments/files/17354746/SETI-Insert.3mf.zip)
[debug_Sun_Oct_13_08_59_07_44501.log.0.zip](https://github.com/user-attachments/files/17354749/debug_Sun_Oct_13_08_59_07_44501.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-10-14:
Can you record a GIF? Thank you. We have already upgraded this feature, please wait for the next version's fix.

shootster, 2024-10-14:
[image]
Here a gif.

Haidiye00, 2024-10-14:
[image]
The internal version should have already been resolved, please wait for the next version~

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00