bambulab/BambuStudio #4988: 设置可变层高后，在使用树状支撑切片时，会增加多余的支撑面

hild597, 2024-10-13:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

A1 mini

### How to reproduce

1、载入模型
2、选择0.2层高
3、开启支撑，选择树状支撑
4、设置可变层高，手动调整
5、切片

### Actual results

会增加多余的支撑面，导致支撑变得非常难拆，甚至直接和模型黏在一起无法拆除
[image]
[image]
[image]

### Expected results

会增加多余的支撑面，导致支撑变得非常难拆，甚至直接和模型黏在一起无法拆除

### Project file & Debug log uploads

[Uploading 旺市牛.zip…]()

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-10-14:
HI [hild597](https://github.com/hild597)：请点击这里上传源文件和截图，当前上传的文件或图片，点击后无法下载成功
[image]

hild597, 2024-10-15:
不发了，我在makerworld上提交工单了，github对于普通人而言太专业了，而且经常访问不了，这个是从Bambu Studio帮助里面的报告问题跳转到githup的，感觉你们需要区分和整合一下你们的机器、makerworld、手机APP和切片软件的售后支持入口，太乱了，看不懂
 

hild
***@***.***

 

 image.png (view on web)

ttt888zzz, 2024-10-16:
抱歉给您带来不好的体验。
1.您提交的工单后，会生成对应的github客诉工单，本次客诉生成地址是：https://github.com/bambulab/BambuStudio/issues/4988
2.我们会按时跟进，先对工单类型打标签：包括bug（特指Studio pc端）， app、marker world、cloud等，在转给对应的团队进行跟进。
[image]
3.我们珍视每一位用户的真实反馈，来优化Studio产品的功能，期待您的反馈，祝您生活愉快