bambulab/BambuStudio #4993: Broken project file after saving and closing

Lil-Diabeetus, 2024-10-13:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu lab A1 mini

### How to reproduce

save project file with several plates
close studio
reopen studio and this project file
"loading of model file failed"
can't open project file

### Actual results

[image]

### Expected results

Open my project file. I've been using this one project file for my warhammer miniature building and have multiple plate setups with presets I would like to restore.

### Project file & Debug log uploads

https://www.dropbox.com/scl/fi/6i8py5tbe492tbxmqig16/Nemestrike.zip?rlkey=y2a08v4pbr25lfg9f5pdfi9id&st=yhilgif0&dl=0

File too big to upload

### Checklist of files to include

- [ ] Log file
- [X] Project file

Lil-Diabeetus, 2024-10-13:
Don't know much about coding, but I might have figured out the issue but not the cause: https://github.com/bambulab/BambuStudio/issues/2541

This post is the same issue, but it seems that my issue is the project
1. Didn't save all my objects, as there are lines within 3dmodel.model/.model.rels that are referencing object models that do not exist in my 3D/objects folder
[image]
2. There are lines that don't match between 3dmodel.model/.model.rels, and this could also be because of point #1 where I am missing some models
[image]

Once again, not a developer or anything. Would like to know if this is the cause of my issue, and the reason that this would happen in the first place.

Haidiye00, 2024-10-14:
Suggest uploading the file, thank you

Haidiye00, 2024-10-15:
[image]
There will be a warning when decompressing
[image]
plate 10 is error

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-04-16 by Haidiye00