bambulab/BambuStudio #4995: Bambu Studio should support "Skip Objects" feature of the Bambu Handy app

altimac, 2024-10-14:
**Is your feature request related to a problem? Please describe.**
In the middle of a multi parts print, one of the part failed to print in the middle and I wanted to skip its printing and only continue printing the other parts. Bambu Handy does support this but not Bambu Studio!

**Describe the solution you'd like**
I would like this cool feature to be integrated directly into Bambu Studio > Device pane

vordenken, 2024-12-21:
+1 
Especially when using a bambulab printer in lan-only mode (execept X1 Carbon) there's no way to use this feature at all.
Please add it to BambuStudio or the printers screen.
I own my P1S for over a year now and never needed this feature till today (shows what a great printer it is) - so learning that this only works when using the app was a real bummer.

OXERY, 2025-02-19:
I also noticed today, that only prints sent via Bambu Studio are being able to skip parts via Bambu Handy.

As I have to print a series of similar prints with many party, my workflow is to reprint directly at the printer - not sending the same Gcode over and over again via Bambu Studio.

Unfortunately, when I start the same print again - from the printer's interface - Bambu handy refuses to skip parts, because the print wasn't sent from Bambu Studio or Handy. WHY?!