bambulab/BambuStudio #5000: 耗材重量计算错误

guoyang37, 2024-10-15:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S

### How to reproduce

切片所有盘后计算耗材重量错误

### Actual results

[image]

### Expected results

切片所有盘后计算耗材重量错误，总计算了2遍

### Project file & Debug log uploads

[这个需要帮我打开查看下.zip](https://github.com/user-attachments/files/17374516/default.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-10-16:
The value of the support material is not displayed correctly, the internal version has been fixed. A new version will be released in the near future.

ttt888zzz, 2024-10-16:
1.Visit website: https://github.com/bambulab/BambuStudio/releases
2.Choose one of the following installation packages to install.
[image]
3.Please download the newest version，and try again.