bambulab/BambuStudio #5003: Moving 3D text after Rotating the part that it's attached to

OpenWorlds1998, 2024-10-15:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

1. Select your part that you will be adding text to; then select Text shape from the tool menu at top
2. Enter your text into the Input text field and click on your part - adding the text to it
3. Set the angle to 270 degrees and click outside of the Text shape editor to exit it
4. Next go into objects and select text_shape (your text) and right-click and select the Change type option
5. From the dropdown select Negative Part and click OK
6. Finally, with the text still selected, click on Move from the tool menu at top
7. Now you will see that entering a number into the X Position field does not produce the desired result

### Actual results

The position number field's value changes to something else after updating (hitting Enter or tabbing)
[image]

### Expected results

The position should use the entered floating point number to set the part to a specific position

### Project file & Debug log uploads

[test_MR241010-16.zip](https://github.com/user-attachments/files/17377755/test_MR241010-16.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2024-10-16:
The new version has been released, please download and use it~

Haidiye00, 2024-10-16:
[image]
[image]
The latest version should be fine

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

OpenWorlds1998, 2024-10-18:
Thank you very much 👍😊