bambulab/BambuStudio #5004: Main screen is black for no reason

DrShweeps, 2024-10-15:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Processeur	AMD Ryzen 7 4800H with Radeon Graphics            2.90 GHz
Mémoire RAM installée	32,0 Go (31,4 Go utilisable)

### Printer

Bambu lab A1 mini

### How to reproduce

1: launch the app

### Actual results

1: i launch the application
2: i see the settings on the left of the screen
3: the main screen i full black, i cant see anything
4: when i click on random locations, after i clicked on "preview", the screen blinks and i can see for less than 1 second
5: the main screen becomes black again
[image]

### Expected results

i want to see what i'm doing

### Project file & Debug log uploads

[Bug.zip](https://github.com/user-attachments/files/17377699/Bug.zip)
[debug_Tue_Oct_15_13_48_28_10600.log.zip](https://github.com/user-attachments/files/17377751/debug_Tue_Oct_15_13_48_28_10600.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-10-16:
1.Visit website: https://github.com/bambulab/BambuStudio/releases
2.Choose one of the following installation packages to install.
[image]
3.Please download the newest version，and try again.

DrShweeps, 2024-10-16:
Thanks for your answer and your work.

I tried both links but the problem is still the same.

guanyun-gudujian, 2024-12-27:
> Thanks for your answer and your work.
> 
> I tried both links but the problem is still the same.
hi, pls help to check if there is a context menu popped after right click in the 3d region, thanks.
like this:
[image]

guanyun-gudujian, 2024-12-27:
@DrShweeps 
hi, pls help to check if there is a context menu popped after right click in the 3d region, thanks.
like this:
[image]

DrShweeps, 2025-01-14:
Hello,
i am sorry for the late answer.
Nothing appeared on the black screen.
i had to totally reboot my computer to unlock the issue.

Thank you very much for you answer.

Have a happy day.

img_v3_02hu_b65bf91c-d328-4937-88a3-7986c044009g.jpg (view on web)<https://github.com/user-attachments/assets/9605d705-b2ff-495c-b0b7-244d2d989a84>

jwcharp, 2025-01-16:
> [@DrShweeps](https://github.com/DrShweeps) hi, pls help to check if there is a context menu popped after right click in the 3d region, thanks. like this: [image]

@guanyun-gudujian This is my issue i get the menu you show

How do I resolve?

guanyun-gudujian, 2025-01-20:
> > [@DrShweeps](https://github.com/DrShweeps) hi, pls help to check if there is a context menu popped after right click in the 3d region, thanks. like this: [image]
> 
> [@guanyun-gudujian](https://github.com/guanyun-gudujian) This is my issue i get the menu you show
> 
> How do I resolve?

can you help me do some debugging stuff?

Haidiye00, 2025-07-22:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~