bambulab/BambuStudio #5006: 1.10 beta testing  - Subnet binding support not binding.

clstr, 2024-10-15:
### Bambu Studio Version

1.10.0

### Where is the application from?

Bambu Lab github releases

### OS version

Manjaro (archlinux)

### Printer

X1C

### How to reproduce

Device Tab
Bind with access code
type in ip address and accesscode

### Actual results

Clicking Connect only yields the following error message:

[image]

### Expected results

Bambu studio should have connected with the device. Computer with bambu studio is on a different subnet and the printer is also on a different subnet.

Printer is in LAN Mode.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

eolder, 2024-10-15:
Seeing similar behavior. P1S running the latest firmware will not connect.

Unlike OP though, my dialog box says "Failed to connect to printer.":
[image]

Taking a look at Wireshark, I can see the printer responds with an ACK with the RST flag set to every packet BambuStudio sends.
Perhaps this feature doesn't work on the release firmware yet and we need to wait for a printer update.

clstr, 2024-10-15:
[image]

eolder, 2024-10-16:
here is the packet capture from my previous comment https://github.com/bambulab/BambuStudio/issues/5006#issuecomment-2415213583

Hope this helps.
[p1s_01.06.01.02_bs_1.10.0.74_subnetbind_anon.pcapng.gz](https://github.com/user-attachments/files/17403011/p1s_01.06.01.02_bs_1.10.0.74_subnetbind_anon.pcapng.gz)

walterwongbbl, 2024-10-24:
@eolder 
The X series printer are currently not supported.

canton7, 2024-10-24:
On seeing this with my A1 mini for what it's worth: "failed to connect to printer" and the printer responding with [ACK, RST]

eolder, 2024-10-24:
> @eolder The X series printer are currently not supported.

Did you mean to type P series? I'm using a P series FYI.

alebre-lithoz, 2024-12-03:
@walterwongbbl is there an estimate on when X1C printers will get the capability to bind via IP-adress?

eolder, 2024-12-10:
I noticed the P1S firmware version `01.07.00.00` was released today so I updated to try it out.

Per the [Bambu Studio 1.10.0 release notes](https://wiki.bambulab.com/en/software/bambu-studio/release/release-note-1-10-0), it should be working now:
>Subnet binding support: [...] (**Supported P-series firmware version:01.07.00.00**,supported A-series firmware version:01.04.00.00,supported X-series firmware:next release version).

Unfortunately, subnet binding is still not working.

I see this now when I configure the printer:
[image]

Linux client Wireshark:
[image]

I also tested this on a windows client, the traffic is the same as the capture I shared previously in this thread.