bambulab/BambuStudio #5007: slicing defect

koob7, 2024-10-15:
### Bambu Studio Version

1.9.7.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Ryzen 5 4600h
rtx2060
16gb ram
lenovo legion 5

### Printer

A1 standard

### How to reproduce

Slicing object from my google drive folder object has some struggle string. They are not a part of stl file. I tried with different slicing preset and effect is still same. Please follow the link to get stl file and screen shots with problem. https://drive.google.com/drive/folders/107qepqyJLgxjc0FJ1j0UhfgfMx6v1d_c?usp=sharing

### Actual results

[image]
[image]

### Expected results

there should not be this extra string
[image]

### Project file & Debug log uploads

[Bryła1.zip](https://github.com/user-attachments/files/17383344/Bryla1.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-10-16:
1.Visit website: https://github.com/bambulab/BambuStudio/releases
2.Choose one of the following installation packages to install.
[image]
3.Try again.

QingZhangBambu, 2024-10-16:
the latest version has fixed that, please try it.

closed 2024-10-17 by ttt888zzz