bambulab/BambuStudio #5008: 1.10.0 beta build failure when system ffmpeg is used

hadess, 2024-10-15:
### Bambu Studio Version

master

### Where is the application from?

Built from source by yourself or others

### OS version

Linux x86-64 Flatpak

### Printer

N/A

### How to reproduce

1.10.0 beta doesn't build if ffmpeg from the system is used.

The build script tries to copy files that only exist when ffmpeg is built as an internal dependency, which isn't needed for most Linux and especially Flatpak builds.

### Actual results

```
set out_dir to CMAKE_CURRENT_BINARY_DIR: /run/build/BambuStudio/build/src
CMake Error at CMakeLists.txt:678 (file):
  file COPY cannot find
  "/run/build/BambuStudio/deps/build/destdir/usr/local/lib/libavcodec.so": No
  such file or directory.
Call Stack (most recent call first):
  src/CMakeLists.txt:228 (bambustudio_copy_sos)
```

### Expected results

it builds

### Project file & Debug log uploads

[stdio-ffmpeg.log](https://github.com/user-attachments/files/17384199/stdio-ffmpeg.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

SaltWei, 2024-10-16:
Known issue @MackBambu will handle it

MackBambu, 2024-10-16:
@hadess Copy the .so file so that it can be easily found when packaging appimage. Is the FFmpeg library already integrated in the Flatpak environment?

hadess, 2024-10-16:
> Is the FFmpeg library already integrated in the Flatpak environment?

It is, we don't need to build it again.

MackBambu, 2024-11-24:
fixed

closed 2024-11-24 by MackBambu