bambulab/BambuStudio #5010: Live View, Please re-install Player on Mac

djblue81, 2024-10-16:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

OS: Mac Sequoia 15.0.1

### Additional system information

Mac Mini M1 16Gb 250Gb

A1 Printer Network Connection = 150Mbit down /60Mbit upload

### Printer

Bambu Lab A1 with AMS lite

### How to reproduce

Sometimes the Player for Liveview dont start.
Error: Player dont Works, Please reinstall Player. Code 2:2
Throubleshoting only for Windows aviable, not for Mac.
On iDevices the Player runns normaly.

[image]

### Actual results

I cant do something.
Mac restart or Studio restart does not help.

### Expected results

what can i do?

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17391122/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2024-10-17:
Please provide log files.
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

bambu123, 2024-10-17:
Sorry. Found logs. Please wait.

djblue81, 2024-10-23:
Hello, is there any news yet?

bambu123, 2024-10-24:
It seems to be a network problem. But was there no problem with the previous Studio version at your place?

djblue81, 2024-10-24:
I only have this version. The image always appears on the mobile Device (iphone), but not on the Mac BambuStudio.

bambu123, 2024-10-24:
Can you help try BambuStudio version 1.9?

djblue81, 2024-10-24:
yes i can try but only on a other older Mac.

djblue81, 2024-10-27:
Look,
on my Mac M1 with 1.9.7.52 it doesn´t make a Picture.
on my old Macbook with 1.9.0.70 it makes a Picture.

[image]

bambu123, 2024-10-28:
According to the log, it is a -22005 error. It may be caused by network changes. Close Studio and try again.

> From: ***@***.***>
> Date:  Sun, Oct 27, 2024, 23:20
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> Look,
on my Mac M1 with 1.9.7.52 it doesn´t make a Picture.
on my old Macbook with 1.9.0.70 it makes a Picture.

djblue81, 2024-10-29:
Closing the studio and reopening it was my first action. Unfortunately without success. 
Please explain to me why it should be a network problem if one computer gets the data from the printer/cloud, the other doesn't.
Mac M1 is connected to the LAN (and parallel WLAN) with 1000mbit and cable, but has no picture. My old Macbook with slower WiFi has a picture. 
Please explain to me, I just want to understand...

bambu123, 2024-10-29:
We judged the network problem through error code -22005. The text prompt on the interface is incorrect. Try using the same network connection method (WiFi).

> From: ***@***.***>
> Date:  Tue, Oct 29, 2024, 15:05
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> Closing the studio and reopening it was my first action. Unfortunately without success.
Please explain to me why it should be a network problem if one computer gets the data from the printer/cloud, the other doesn't.
Mac M1 is connected to the LAN (and parallel WLAN) with 1000mbit and cable, but has no picture. My old Macbook with slower WiFi has a picture.
Please explain to me, I just want to understand...

peteramelang, 2024-11-10:
@bambu123 

I have the same issue on my Mac after updating to MacOS 15. I have deleted everything from Application Support and reinstalled BambuStudio and the network driver.

I can watch the video feed or access the SD card from my Mac.
No issues with access through the mobile app.

bambu123, 2024-11-11:
@peteramelang<https://github.com/peteramelang> 
Can you access the files of the printer sd card through studio? In case of liveview failure. In any case, please provide us with the right-click log.

> From: ***@***.***>
> Date:  Mon, Nov 11, 2024, 05:20
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> @bambu123<https://github.com/bambu123>
> I have the same issue on my Mac after updating to MacOS 15. I have deleted everything from Application Support and reinstalled BambuStudio and the network driver.
> I can watch the video feed or access the SD card from my Mac.
No issues with access through the mobile app.

bambu123, 2024-11-11:
Can you help me see if this is useful for you?

https://youtrack.jetbrains.com/articles/SUPPORT-A-564/macOS-15-Local-Network-Privacy-blocks-IDE-connections-to-local-resources-databases-app-servers-etc-unless-explicitly-permitted
[image: https://youtrack.jetbrains.com/api/files/334-61502?sign=MTczMTU0MjQwMDAwMHwxMS0xfDMzNC02MTUwMnw2OVNqc0F4VXM4OFMya1plQjUzT3IxZ1NITjFrTmRXX2FmSDRIdmpvdk9JDQo&updated=1727193593765]

> From: ***@***.***>
> Date:  Mon, Nov 11, 2024, 05:20
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> @bambu123<https://github.com/bambu123>
> I have the same issue on my Mac after updating to MacOS 15. I have deleted everything from Application Support and reinstalled BambuStudio and the network driver.
> I can watch the video feed or access the SD card from my Mac.
No issues with access through the mobile app.

peteramelang, 2024-11-12:
@bambu123, I also have it turned on. Just updated to v01.10.00.89, but no change either. I can not access the SD card.
My log: [debug_Tue_Nov_12_22_17_54_31986.log.0.zip](https://github.com/user-attachments/files/17722931/debug_Tue_Nov_12_22_17_54_31986.log.0.zip)

miguesan01, 2024-11-29:
> Can you help me see if this is useful for you? [...]

Turning off the and on the toggle button works at the moment, but next time I open Bambu I have to do it again, it's annoying. The second solution does nothing in my Mac since  firewall is disabled.

marcosgeorgopoulos, 2024-12-18:
I have the same problem and have raised a ticket with Bambu Labs support

djblue81, 2025-01-17:
Is there a Solution being developed??

djblue81, 2025-02-17:
No answer is also an answer

miguesan01, 2025-02-17:
> No answer is also an answer. Its so easy, you take a finger and press the letters Y E S or N O on the keyboard and then "Comment".

It seems to be an macOS issue, not only with Bambu studio but also with other apps that use local network. On my mac it was solved updating to the last OS version.