bambulab/BambuStudio #5011: No video from the printer camera

roland-blaser, 2024-10-16:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.0.1

### Additional system information

Mac Pro 2023 (Apple Silicon)

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Yesterday the video has been sent from the printer to the Bambu Studio app on my Mac (within same WLAN network in my home). In between the printer had been switch off (via power switch at the back). Today I started another print job but the video is not working anymore. The same for the Bambu app on my iPhone via the internet: no video.
Important is: All other data (e.g. printing progress, amounts of printed layers, remaining time) are coming through, but no video. Suggest software bug within the respective firmware.

### Actual results

Printer video not sent anymore to Bambu Studio on Mac and Bambu App on iPhone.

### Expected results

Video should be working as before.

### Project file & Debug log uploads

no additional information needed, see above.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-10-17:
Please provide log files.
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

roland-blaser, 2024-10-17:
[Bambu Log.zip](https://github.com/user-attachments/files/17409744/Bambu.Log.zip)
It wasn't easy to find the log files for the date/time where the problem with missing video happened. Most probably it is the file that covers the time in between the two others. Important to know: After switching the X1 printer off/on the camera reappeared within Bambu Studio (Mac) and within the Bambu iOS App on the iPhone. Strange anyway...