bambulab/BambuStudio #5012: German grammatical error in Device->Update->Printer

T-electroJo, 2024-10-16:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

0. German language chosen
1. Go to the device tab
2. Go to the update sidetab

### Actual results

Above the printer, after the printers's name, there is "(Drucken)"

### Expected results

Above the printer, after the printers's name, there should be **(Beschäftigt)**, **(Druckt)** or **(Im Druckvorgang)**

### Project file & Debug log uploads

None needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

T-electroJo, 2024-10-16:
Its not like a real big grammatical error, but it will be pronounced differently here in Germany so i thought i just bring it up here to just let it look nicer ^^.

LiZ-Li-BBL, 2024-10-23:
Thank you for your suggestion. It will get modified after confirmation.