bambulab/BambuStudio #5018: Target temps no longer displayed

sprior, 2024-10-17:
### Bambu Studio Version

1.10.0.74

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu X1C

### How to reproduce

Start a print, look in Bambu Studio at the printer status page, notice that the actual nozzle/bed/chamber temps are displayed but the target temps are all missing.
[image]

### Actual results

Target temps aren't displayed next to the actual temps

### Expected results

Target temps should be displayed next to the actual temps

### Project file & Debug log uploads

[New Compressed (zipped) Folder.zip](https://github.com/user-attachments/files/17404882/New.Compressed.zipped.Folder.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2024-10-17:
@sprior 
We just fixed it and it will be included in the next version.

HomerSimpy, 2024-11-13:
made it worse 
[image]