bambulab/BambuStudio #5019: SVG Import just imports outline, none of the inside in Bambu Studio Beta 1.10 for MAC (1.09.... working ok)

MauricioMarz, 2024-10-17:
### Bambu Studio Version

1.10 Beta

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS 15 and 15.1

### Additional system information

MacBook Air M2

### Printer

A1

### How to reproduce

Importing any svg just import 1 object (outline), other versions import all objects of svg 

### Actual results

.

### Expected results

.

### Project file & Debug log uploads

.

### Checklist of files to include

- [X] Log file
- [ ] Project file

Haidiye00, 2024-10-17:
Thank you for your discovery. We will fix it as soon as possible~

Haidiye00, 2025-01-11:
Hello, the new beta version has been released. Welcome to download and experience it~
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-11 by Haidiye00