bambulab/BambuStudio #5023: By-object distance warning should trigger on objects, not clearance radii

myklemu, 2024-10-17:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

X1C 

### How to reproduce

- Add two or more objects to build plate
- Set to "by object" print sequence
- Position the objects so that the circles overlap, but do not intersect or include another object

### Actual results

Warning message "File_name.STL is too close to others, and collisions may be caused". 
Slicer will not slice, button greyed out.

### Expected results

[image]

As long as extruder_clearance_max_radius does not include another object, it should still be okay that it overlaps with another clearance radius. The distance is mutual, thus no collision should occur. The warning should only show if the neighbouring  _object_ is inside the clearance radii. This will allow printing far more objects per plate.

### Project file & Debug log uploads

Not applicable.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-10-18:
To ensure that each piece of printing avoids collision, this is currently the design. Currently, our latest software version is 1.10beta, which you can download and use~

closed 2024-10-18 by Haidiye00

myklemu, 2024-10-21:
> To ensure that each piece of printing avoids collision, this is currently the design. Currently, our latest software version is 1.10beta, which you can download and use~

@Haidiye00 
I downloaded 1.10 beta, it looks the same / not fixed. Could you double check that you understand the issue, please? 

- Boundary crosses another **boundary** is OK! Will print fine. Please remove the warning and enable slicing.
- Boundary includes another **object** is _not_ OK, show warning and disable slicing (as before).