bambulab/BambuStudio #5024: Feature Request - Fuzzy Skin Drawing and Seam Drawing Options

thomasvdsterre122, 2024-10-17:
Dear Bambulab Software Team,

I hope this message finds you well. I would like to request two new features for the drawing tool that could greatly improve the user experience:

1. Fuzzy Skin Drawing: Just like "Support Drawing" allows users to visualize and create supports, a "Fuzzy Skin Drawing" option would enable users to add a rough texture to objects. This feature would also allow users to keep the inside of objects smooth while applying a fuzzy texture to the outside. It would be especially useful for creating objects with a rough exterior and smooth inner surfaces.

2. Seam Drawing: Additionally, an option to visualize and draw seams on models would greatly help optimize the design for 3D printing. By being able to see and adjust where the seams appear, users could improve the structural integrity and visual aspect of their models.

magnusmyklebost, 2024-11-27:
Please just add the same functions as the paint tool, so you could fill faces with fuzzy skin.

Please implement the stuff this genius has done:
https://www.youtube.com/watch?v=cNkHfydnUCI

magnusmyklebost, 2024-11-27:
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1 Prusa have added the features...

cemkalyoncu, 2025-02-28:
This feature is important in cases where fuzzy skin is applied to a functional part. Sometimes one of the walls needs to be smooth. Selective fuzzy skin can help in those cases.

exeljb, 2025-03-27:
I'd definitely like to see this feature added to Bambu studio especially since prusa finally added it to their slicer