bambulab/BambuStudio #5028: Finish Time not being shown

EricDimitri, 2024-10-17:
### Bambu Studio Version

Beta 1.10.0.74

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab P1S AMS

### How to reproduce

Print something, then check the Device Tab. "Finish Time:" label is there, but no time being shown after that

### Actual results

Finish time not being Shown

### Expected results

Finish time should be shown after the label

### Project file & Debug log uploads

Project file
[PROJECT.zip](https://github.com/user-attachments/files/17424869/PROJECT.zip)
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

MartinChan3, 2024-10-30:
[1.10 beta2](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81) has solved this problem.

EricDimitri, 2024-10-30:
Thanks a lot!!!

EricDimitri, 2024-10-30:
I just installed the latest beta version, and still cant see  the Finish time. Please tell me what do you need from my side. Thanks!!!

DanBao-Bambu, 2024-11-04:
We apologize for any inconvenience caused and have added logs to the code to track this issue. But the log level needs to be set to trace. Here are the methods:
Enter the Configuration folder
[image]
Use a text editor to edit BambuStudio.conf, locate the 'severity level' field, and change it to 'trace'
[image]
Then open BambuStudio and when the finish_time does not display, copy the log and send it to me as an attachment. Thank you very much.

EricDimitri, 2024-11-04:
Ok @DanBao-Bambu , will do, but keep in mind that it NEVER shows the finish_time to me. Anyway, I will send you logs in trace lvl ASAP. Thanks a lot!!

EricDimitri, 2024-11-04:
[debug_Mon_Nov_04_08_44_01_20740.log.zip](https://github.com/user-attachments/files/17617688/debug_Mon_Nov_04_08_44_01_20740.log.zip)
Is this the log you need? Let me know please. Thanks!!!

EricDimitri, 2024-11-04:
Weird thing, right after I modified the log lvl in the CFG file, sent to print, and appeared... I dont know if it is related or what, but.... Really weird....

EricDimitri, 2024-11-04:
*** Update ***
It works just sometimes... sometimes ut shows, sometimes it doesnt...  I dont know related to what.

MartinChan3, 2024-11-05:
Can you offer a video or gif to show in which situation it doesn't work ? We should know more details...

EricDimitri, 2024-11-05:
As far as I tested, some prints show it, some prints dont show it. How a video is helpful? I can record one but, You will see it not being shown, as in the Screenshot :)

MartinChan3, 2024-11-05:
@EricDimitri I had print  with my A1 and P1P by Studio V1.10 Beta2, and I didn't figure out in which situation this could happen...So I asked the video so I can test like you.

EricDimitri, 2024-11-05:
@MartinChan3 I am doing nothing weird, just send to print from Studio... Sometimes it shows, sometimers it doesnt. Couldnt figure oput the pattern for it. Next time It doesnt show it, will share the project for you to test.

DanBao-Bambu, 2024-11-06:
[image]
> Is this the log you need? Let me know please. Thanks!!!

@EricDimitri  I have checked your log and did not find any abnormalities, which is strange. Perhaps this log records when you were displaying the finish time normally?

EricDimitri, 2024-11-08:
I saw the same... But no. I deleted log, then send to print, and then after checking it is not being shown on the screen, saved the log... Weird...

EricDimitri, 2024-11-10:
[image]
Look at this. Only thing I might think about is that it will finish the next day? But something is not ok there :)

HomerSimpy, 2024-11-13:
1.10.0.89 shows line through on higher Rez monitors
[image]

RosomakPL, 2024-11-28:
Same here but version 1.10.1.50
[image]

Rich-b1, 2024-12-08:
I'm having this same problem with Studio 1.10.1.50
Finish Time under the Progress bar will not display a time of day. 
But it will change to "Finished" when the print is done.

I have TWO computers, but only ONE X1C printer.

1. On my PC which is running Windows 10, Finish time will NOT display.
2. On my LAPTOP running Windows 11, with the same version of Studio. finish time displays OK!

I tried several things such as:

1. Connect both computers via Ethernet (previously the laptop was via WiFi but the PC was via Ethernet).
2. Uninstalled Studio, rebooted PC, Re-installed Studio.
3. Logged out of my Bambu account, and then logged in again.

Nothing seems to allow the PC to display the Finish Time.

Please advise if I should submit some logs or if this is a known issue being actively worked on already? I don't want to duplicate your efforts.

Rich-b1, 2024-12-09:
Please see my edited post. I found that by changing my Windows Screen Scale from 100% to another value (such at 125%) works around the issue and allows the Finish time to be displayed.

EricDimitri, 2024-12-13:
> Please see my edited post. I found that by changing my Windows Screen Scale from 100% to another value (such as 125%) works around the issue and allows the Finish time to be displayed.

Thats cool but, Just to see that number, change the scale of the whole monitor, is not worth it. It has to be fixed "right". anyway, thanks for the workaround!! And Happy New Year!!!

JohnSmithCYM, 2024-12-14:
@Rich-b1 Thanks for your test 🐱 and we will check if it works on 1080P 100% in our internal version.

Rich-b1, 2024-12-14:
@DanBao-Bambu @JohnSmithCYM I just purchased a new monitor. This can have a resolution of **5120 x 1440**. Even with this large resolution, if the screen "Scale and layout" is set to **100%**, Finish time is **NOT** displayed!

If I choose other Scale settings in Windows (125%, 150%, 175%, 200%, 225%), then Finish time **WILL** display ok!

So it seems it's not that the screen layout doesn't have enough room for the Finish time to be displayed, but something about the Scale and layout being at 100% that causes the problem.

Note that I'm currently using Windows 10 and have I'm using a nvidea GFORCE GTX980Ti GPU card.

I hope this helps you! If you want me to do any more testing, please let me know!

Bladevane, 2024-12-23:
I totally agree with the above comments - the user should not have to change the screen display to see the remaining time. Us users need a fix from Bambu Labs, not a Windows workaround.

Rich-b1, 2024-12-23:
> I totally agree with the above comments - the user should not have to change the screen display to see the remaining time. Us users need a fix from Bambu Labs, not a Windows workaround.

Personally, I don't think anyone is suggesting using a work around. But the work around lets you see the Finish time until they release a fixed version of Studio.

It sure would be nice if someone from BL would acknowledge that they have duplicated this issue, and the fix will be in the next release. But I'm not sure if the previous comments show that they understand the root cause yet.

Bladevane, 2024-12-24:
Thanks for the reply. Yes, it would be appreciated if BL could at least acknowledge there is an issue which needs to be addressed. Maybe they are looking into it, who knows??

Rich-b1, 2025-01-25:
I also found this is occurring in Bambu Studio Beta Version 1.10.2.64 and on Windows 11 (previous for me was Windows 10).

If the Windows Screen Scale is set to 100%, then Finish Time is NOT displayed:

[image]

If the Windows Screen Scale is set to 125$, then Finish Time is displayed OK:

[image]

LiamDroog, 2025-04-02:
Seconding this occurring on 2.0.0.95. Windows 10,  2560x1440 monitor. One might think that BBL could fix such an easy bug in six months...

[image]

EricDimitri, 2025-07-17:
@DanBao-Bambu  Loos like this was fixed on the beta, and they didnt mention this bug, that was reported way before hahahahahahaha

closed 2025-08-19 by JohnSmithCYM