bambulab/BambuStudio #5029: Build plate texture image acting odd (Creality K1)

just-trey, 2024-10-18:
### Bambu Studio Version

1.10.0.74

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Creality K1

### How to reproduce

1. Add a Creality K1 printer .04 nozzle
2. Add network connection to the printer and save as a new printer

### Actual results

The bambu build plate texture shows up 

.

### Expected results

The creality build plate texture should appear. 
[image]

### Project file & Debug log uploads

not crasing and no project open

### Checklist of files to include

- [ ] Log file
- [X] Project file

ttt888zzz, 2024-10-18:
HI [just-trey](https://github.com/just-trey)：Thank you for your feedback. Could you Please Provide a video of the operation to help solve the issue?

just-trey, 2024-10-18:
@ttt888zzz Let me know if this is good for you. 

https://github.com/user-attachments/assets/123a8145-859e-45f5-bcd9-2d0ebd57fce7

Haidiye00, 2024-10-18:
Thank you for your discovery. We have identified and resolved the issue, thank you

ttt888zzz, 2024-10-18:
HI [just-trey](https://github.com/just-trey)：This video helped a lot and the internal version has been fixed. The next version will be available soon.

Haidiye00, 2024-10-29:
Hello, 1.10 beta2 has been released. Please download and use it. Thank you~

just-trey, 2024-11-15:
@Haidiye00  and @ttt888zzz  I am seeing this issue again with the final release. Can one of you reopen this?

Haidiye00, 2024-11-15:
Hello, do you think you can use the latest version https://github.com/bambulab/BambuStudio/releases Can you help us record a video? Thank you. Let's double check~

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-04-16 by Haidiye00