bambulab/BambuStudio #5033: No video connection and time-lapse video at my p1s

fito1012, 2024-10-18:
### Bambu Studio Version

01.09.07.52

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS

### Additional system information

MacBook Pro M1 pro

### Printer

P1S

### How to reproduce

Try to open Timelaps video

### Actual results

[image]
[image]

### Expected results

See time-lapse videos or see realtime camera video

### Project file & Debug log uploads

Fri Oct 18 19:01:04:test storage upgrade start…
Fri Oct 18 19:01:04:[test_oss_upgrade]: url=http://upgrade-file.bambulab.com
Fri Oct 18 19:01:05:test storage start…
Fri Oct 18 19:01:05:[test_oss]: url=http://upload-file.bambulab.com
Fri Oct 18 19:01:05:test storage upgrade failed
Fri Oct 18 19:01:05:status=404, body={„error_msg“:“404 Route Not Found“}
, error=
Fri Oct 18 19:01:06:test storage failed
Fri Oct 18 19:01:06:status=404, body={„error_msg“:“404 Route Not Found“}
, error=
Fri Oct 18 19:01:06:test http start…
Fri Oct 18 19:01:06:test http ok
Fri Oct 18 19:01:07:test bing start…
Fri Oct 18 19:01:07:[test_bing]: url=http://www.bing.com/
Fri Oct 18 19:01:07:test bing ip resolved = 184.86.251.22
Fri Oct 18 19:01:07:test bing ok
Fri Oct 18 19:01:07:test bambulab start…
Fri Oct 18 19:01:07:[test_bambulab]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?name=slicer&version=01.09.07.52&guide_version=00.00.00.01
Fri Oct 18 19:01:08:test bambulab ip resolved = 104.18.35.41
Fri Oct 18 19:01:08:test bambulab ok
Fri Oct 18 19:01:10:test plugin download start…
Fri Oct 18 19:01:10:[test_plugin_download]: downloadurl=https://public-cdn.bblmw.com/upgrade/studio/plugins/01.09.07.01/1c014710f0/mac_01.09.07.01.zip
Fri Oct 18 19:01:10:downloading 5%
Fri Oct 18 19:01:10:downloading 10%
Fri Oct 18 19:01:10:downloading 15%
Fri Oct 18 19:01:10:test storage download start…
Fri Oct 18 19:01:10:[test_oss_download]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.09.07.00
Fri Oct 18 19:01:10:downloading 20%
Fri Oct 18 19:01:10:downloading 25%
Fri Oct 18 19:01:10:downloading 30%
Fri Oct 18 19:01:11:downloading 35%
Fri Oct 18 19:01:11:downloading 40%
Fri Oct 18 19:01:11:downloading 45%
Fri Oct 18 19:01:11:downloading 50%
Fri Oct 18 19:01:11:downloading 55%
Fri Oct 18 19:01:11:downloading 60%
Fri Oct 18 19:01:11:downloading 65%
Fri Oct 18 19:01:11:downloading 70%
Fri Oct 18 19:01:11:downloading 75%
Fri Oct 18 19:01:11:downloading 80%
Fri Oct 18 19:01:11:test storage upgrade start…
Fri Oct 18 19:01:11:[test_oss_upgrade]: url=http://upgrade-file.bambulab.com
Fri Oct 18 19:01:11:downloading 85%
Fri Oct 18 19:01:11:downloading 90%
Fri Oct 18 19:01:11:downloading 95%
Fri Oct 18 19:01:11:downloading 100%
Fri Oct 18 19:01:11:test plugin download ok
Fri Oct 18 19:01:11:downloading 10%
Fri Oct 18 19:01:11:downloading 20%
Fri Oct 18 19:01:12:test storage upgrade failed
Fri Oct 18 19:01:12:status=404, body={„error_msg“:“404 Route Not Found“}
, error=
Fri Oct 18 19:01:12:downloading 30%
Fri Oct 18 19:01:12:downloading 40%
Fri Oct 18 19:01:12:downloading 50%
Fri Oct 18 19:01:12:downloading 60%
Fri Oct 18 19:01:12:downloading 70%
Fri Oct 18 19:01:12:downloading 80%
Fri Oct 18 19:01:12:downloading 90%
Fri Oct 18 19:01:12:downloading 100%
Fri Oct 18 19:01:12:test storage download ok

### Checklist of files to include

- [X] Log file
- [ ] Project file

ttt888zzz, 2024-10-21:
HI [fito1012](https://github.com/fito1012): Could you please Provide the log file？
1.open Studio and click the menu
[image]
2.Find the log for the current time.
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log
3.upload the log file
[image]

supernerd11111, 2024-11-15:
Same issue
Bambu Version: 10.10.1.5
Mac M1 Pro
Device cam doesn't work, timelapse doesn't work, models doesn't work. All result in the screenshots shown exact