bambulab/BambuStudio #5034: 1.10.0 beta build failure

fr3ak2402, 2024-10-18:
### Bambu Studio Version

master branch

### Where is the application from?

Built from source by yourself or others

### OS version

Windows 11

### Printer

n/A

### How to reproduce

erere

### Actual results

erer

### Expected results

it builds

### Project file & Debug log uploads

n/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Sebazzz, 2024-10-18:
Also running into this issue. I resolved this by placing extracting [a pre-build version](https://stackoverflow.com/a/1711338) into the BambuStudio_deps directory, however there seem to be further compatiblity issues with Visual Studio 2022.

fr3ak2402, 2024-10-18:
@Sebazzz Thanks for the quick answer.

Did you unpack the pre-built libs directly into the BambuLab_deps folder or into a subdirectory like usr/local?

I am currently compiling everything with VS2019, as I also had problems with VS2022. Somehow building OpenSSL under VS2022 never worked because of not enough environment memory.

Sebazzz, 2024-10-18:
> I am currently compiling everything with VS2019, as I also had problems with VS2022. Somehow building OpenSSL under VS2022 never worked because of not enough environment memory.

I actually had no problems with that, but at some point I run into linking errors with curl: 

```
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_mime_init [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_mime_free [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_mime_addpart [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_mime_name [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_mime_filename [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_mime_type [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_mime_data [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_mime_filedata [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_formadd [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_formfree [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_easy_escape [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_easy_unescape [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_free [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(OctoPrint.obj) : error LNK2001: unresolved external symbol curl_free [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_global_init [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_global_cleanup [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_slist_append [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_slist_free_all [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_easy_strerror [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_easy_init [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_easy_setopt [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_easy_perform [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_easy_cleanup [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(Http.obj) : error LNK2001: unresolved external symbol curl_easy_getinfo [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(OctoPrint.obj) : error LNK2001: unresolved external symbol curl_url [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(OctoPrint.obj) : error LNK2001: unresolved external symbol curl_url_cleanup [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(OctoPrint.obj) : error LNK2001: unresolved external symbol curl_url_get [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
libslic3r_gui.lib(OctoPrint.obj) : error LNK2001: unresolved external symbol curl_url_set [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
P:\Dev\BambuStudio\build\src\Release\BambuStudio.dll : fatal error LNK1120: 27 unresolved externals [P:\Dev\BambuStudio\build\src\BambuStudio.vcxproj]
```

lanewei120, 2024-10-19:
we need to install pkg-config for windows building now

just updated it to the wiki :
pkg-config from [sourceforge](https://sourceforge.net/projects/pkgconfiglite/files/latest/download)

fr3ak2402, 2024-10-19:
@lanewei120 Thank you for your answer. 

I still don't know where to install the pkg-config. 
I downloaded the zip and there is no installer in there, but a folder. 
Where do I have to copy the files?

fr3ak2402, 2024-10-19:
I got it right.
Just had to copy the files of the folder from pkg-config into the respective folder of the deps

closed 2024-10-20 by fr3ak2402