bambulab/BambuStudio #5036: black screen

benjaa20, 2024-10-18:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

bambu lab a1

### How to reproduce

1. open bambu studio
2. go to prepare or download a model
3. the screen is black

### Actual results

[image]

### Expected results

the printing plate and model is normally visible

### Project file & Debug log uploads

[Ironman_SupportFree.zip](https://github.com/user-attachments/files/17440973/Ironman_SupportFree.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2024-10-28:
Hello, could you please provide the graphics card model of your computer? Thank you~

guanyun-gudujian, 2024-12-27:
> ### Bambu Studio Version
> 1.9.7.52
> 
> ### Where is the application from?
> Bambu Lab Official website
> 
> ### OS version
> windows 11
> 
> ### Additional system information
> _No response_
> 
> ### Printer
> bambu lab a1
> 
> ### How to reproduce
> 1. open bambu studio
> 2. go to prepare or download a model
> 3. the screen is black
> 
> ### Actual results
> ![imagen_2024-10-18_183624461](https://private-user-images.githubusercontent.com/185559305/378017867-44d2327f-30eb-408b-ba77-f1b01a92e1da.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUyODIzNzUsIm5iZiI6MTczNTI4MjA3NSwicGF0aCI6Ii8xODU1NTkzMDUvMzc4MDE3ODY3LTQ0ZDIzMjdmLTMwZWItNDA4Yi1iYTc3LWYxYjAxYTkyZTFkYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyN1QwNjQ3NTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kNTY1YTM3N2NjZDI5M2I5MWE3NmUzNDdjZTg1ODliNTE4Y2NkMmM3MGE2MTA3MmY3NDNmMGNmZjFkOTBhMDY3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.f8Xdf-kvfOl0HuxCW5t4Pc67BCLVkqvI20-jNehqWTw)
> 
> ### Expected results
> the printing plate and model is normally visible
> 
> ### Project file & Debug log uploads
> [Ironman_SupportFree.zip](https://github.com/user-attachments/files/17440973/Ironman_SupportFree.zip)
> 
> ### Checklist of files to include
> * [ ]  Log file
> * [x]  Project file

hi, pls help to check if there is a context menu popped after right click in the 3d region, thanks.
like this:
[image]

MikhailovAnton, 2025-03-23:
You can fix this by using discrete GPU / Nvidea GPU.

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95