bambulab/BambuStudio #5037: Access violation while slicing file (0xC0000005) - BambuStudio!BRepExtrema_SelfIntersection

Sebazzz, 2024-10-19:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 - 24H2

### Additional system information

CPU: AMD Ryzen 3700X
GFX: Nvidia RTX3090

### Printer

Bambu Lab X1C

### How to reproduce

1. Load the file
2. Slice it

### Actual results

Crash

### Expected results

Preview should be shown

### Project file & Debug log uploads

Project: [Lulkoek-tegel_Front_100x100_S_MC.zip](https://github.com/user-attachments/files/17442761/Lulkoek-tegel_Front_100x100_S_MC.zip)

I ran BambuStudio in WinDBG, and turned up with this analysis:

```
(5f5c.577c): Access violation - code c0000005 (first chance)
First chance exceptions are reported before any exception handling.
This exception may be expected and handled.
BambuStudio!BRepExtrema_SelfIntersection::`default constructor closure'+0x293555:
00007ffe`f42c4705 8b4218          mov     eax,dword ptr [rdx+18h] ds:00000000`00000018=????????
0:022> !analyze -v
*******************************************************************************
*                                                                             *
*                        Exception Analysis                                   *
*                                                                             *
*******************************************************************************

KEY_VALUES_STRING: 1

    Key  : AV.Dereference
    Value: NullClassPtr

    Key  : AV.Fault
    Value: Read

    Key  : Analysis.CPU.mSec
    Value: 1187

    Key  : Analysis.Elapsed.mSec
    Value: 17244

    Key  : Analysis.IO.Other.Mb
    Value: 105

    Key  : Analysis.IO.Read.Mb
    Value: 2

    Key  : Analysis.IO.Write.Mb
    Value: 156

    Key  : Analysis.Init.CPU.mSec
    Value: 1546

    Key  : Analysis.Init.Elapsed.mSec
    Value: 304628

    Key  : Analysis.Memory.CommitPeak.Mb
    Value: 183

    Key  : Analysis.Version.DbgEng
    Value: 10.0.27704.1001

    Key  : Analysis.Version.Description
    Value: 10.2408.27.01 amd64fre

    Key  : Analysis.Version.Ext
    Value: 1.2408.27.1

    Key  : Failure.Bucket
    Value: NULL_CLASS_PTR_READ_c0000005_BambuStudio.dll!Unknown

    Key  : Failure.Hash
    Value: {6d5aadbf-9bdb-600a-a1fb-088c78a6913d}

    Key  : Timeline.OS.Boot.DeltaSec
    Value: 383

    Key  : Timeline.Process.Start.DeltaSec
    Value: 304

    Key  : WER.OS.Branch
    Value: ge_release

    Key  : WER.OS.Version
    Value: 10.0.26100.1

    Key  : WER.Process.Version
    Value: 1.0.0.0

NTGLOBALFLAG:  70

APPLICATION_VERIFIER_FLAGS:  0

EXCEPTION_RECORD:  (.exr -1)
ExceptionAddress: 00007ffef42c4705 (BambuStudio!BRepExtrema_SelfIntersection::`default constructor closure'+0x0000000000293555)
   ExceptionCode: c0000005 (Access violation)
  ExceptionFlags: 00000000
NumberParameters: 2
   Parameter[0]: 0000000000000000
   Parameter[1]: 0000000000000018
Attempt to read from address 0000000000000018

FAULTING_THREAD:  0000577c

PROCESS_NAME:  bambu-studio.exe

READ_ADDRESS:  0000000000000018 

ERROR_CODE: (NTSTATUS) 0xc0000005 - The instruction at 0x%p referenced memory at 0x%p. The memory could not be %s.

EXCEPTION_CODE_STR:  c0000005

EXCEPTION_PARAMETER1:  0000000000000000

EXCEPTION_PARAMETER2:  0000000000000018

STACK_TEXT:  
0000009e`9e63ec00 00007ffe`f41bc4b1     : 00000000`00000000 000001b7`5b3639d0 00000000`00000001 00000000`00000000 : BambuStudio!BRepExtrema_SelfIntersection::`default constructor closure'+0x293555
0000009e`9e63eca0 00007ffe`f3febfa5     : 2aaaaaaa`aaaaaaab 0000009e`9e63ee38 000001b7`5ada01b0 00007ffe`f3d6219a : BambuStudio!BRepExtrema_SelfIntersection::`default constructor closure'+0x18b301
0000009e`9e63ee00 00007ffe`f4333f91     : 000001b7`5b3639d0 0000009e`9e63f240 00000000`00000001 00000000`00000000 : BambuStudio!cereal::detail::StaticObject<cereal::detail::bind_to_archives<Slic3r::DynamicPrintConfig,cereal::detail::`anonymous namespace'::polymorphic_binding_tag> >::instantiate+0x226a45
0000009e`9e63f080 00007ffe`f420f018     : 000001b7`3f13d8f0 000001b7`3f13d8f0 000001b7`3f0620f0 000001b7`3ee994a0 : BambuStudio!BRepExtrema_SelfIntersection::`default constructor closure'+0x302de1
0000009e`9e63f530 00007ffe`f415fd31     : 00000000`00000000 000001b7`3ee994a0 000001b7`3ee99510 000001b7`3ee99528 : BambuStudio!BRepExtrema_SelfIntersection::`default constructor closure'+0x1dde68
0000009e`9e63f7a0 00007ffe`f412d348     : 02fc75e2`02faef49 00000000`00000012 000001b7`31d6f8c0 000001b7`31d6f880 : BambuStudio!BRepExtrema_SelfIntersection::`default constructor closure'+0x12eb81
0000009e`9e63f950 00007ffe`f40fd008     : 000001b7`31d6f880 000001b7`32069e08 000001b7`3f490e00 000001b7`32069e08 : BambuStudio!BRepExtrema_SelfIntersection::`default constructor closure'+0xfc198
0000009e`9e63f9a0 00007ffe`f41247e9     : 000001b7`31d6f880 00000000`00000000 000001b7`31d6f880 000001b7`3f3ff280 : BambuStudio!BRepExtrema_SelfIntersection::`default constructor closure'+0xcbe58
0000009e`9e63fb10 00007ffe`f3c206a0     : 000001b7`31b48000 000001b7`31b47e00 00000000`00000000 0000009e`9e63fc80 : BambuStudio!BRepExtrema_SelfIntersection::`default constructor closure'+0xf3639
0000009e`9e63fb60 00007ffe`f3c1bcc2     : 000001b7`32069e00 000001b7`31b47e01 0000009e`9e63fc80 00000000`00001f01 : BambuStudio+0x2e06a0
0000009e`9e63fc60 00007ffe`f3c2701c     : 000001b7`31b47e00 00000000`00000000 000001b7`31b47e00 00000000`00000000 : BambuStudio+0x2dbcc2
0000009e`9e63fcc0 00007ffe`f3c2e4e7     : 000001b7`31a561c0 00000000`00000000 000001b7`31b47e00 000001b7`31a63b00 : BambuStudio+0x2e701c
0000009e`9e63fd00 00007ffe`f3c2e3ab     : 00000000`00000000 00000000`00000008 000001b7`31a561c0 00000000`00000000 : BambuStudio+0x2ee4e7
0000009e`9e63fd40 00007fff`da334ea0     : 000001b7`31a561c0 00000000`00000000 00000000`00000000 00000000`00000000 : BambuStudio+0x2ee3ab
0000009e`9e64f820 00007fff`db1edbe7     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ucrtbase!thread_start<unsigned int (__cdecl*)(void *),1>+0x30
0000009e`9e64f850 00007fff`dc6a5a4c     : 00000000`00000000 00000000`00000000 000004f0`fffffb30 000004d0`fffffb30 : KERNEL32!BaseThreadInitThunk+0x17
0000009e`9e64f880 00000000`00000000     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ntdll!RtlUserThreadStart+0x2c

SYMBOL_NAME:  BambuStudio+293555

MODULE_NAME: BambuStudio

IMAGE_NAME:  BambuStudio.dll

STACK_COMMAND:  ~22s ; .cxr ; kb

FAILURE_BUCKET_ID:  NULL_CLASS_PTR_READ_c0000005_BambuStudio.dll!Unknown

OS_VERSION:  10.0.26100.1

BUILDLAB_STR:  ge_release

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {6d5aadbf-9bdb-600a-a1fb-088c78a6913d}

Followup:     MachineOwner

```

### Checklist of files to include

- [X] Log file
- [X] Project file

Sebazzz, 2024-10-19:
(I would have debugged it myself, but I'm running into curl linking issues as mentioned in my comment in #5034)

zhimin-zeng-bambulab, 2024-10-24:
Thanks for reporting this issue, we will fix it in the next version.

MartinChan3, 2024-10-30:
[1.10 beta2](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81) has solved this problem.

closed 2024-10-30 by zhimin-zeng-bambulab