bambulab/BambuStudio #5038: Prepare and Preview graphical regions don't render under Ubuntu 24.10

appositeit, 2024-10-19:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.10

### Additional system information

System:
  Host: mie Kernel: 6.11.0-8-generic arch: x86_64 bits: 64
  Desktop: GNOME v: 47.0 Distro: Ubuntu 24.10 (Oracular Oriole)
Machine:
  Type: Desktop System: Micro-Star product: MS-7C37 v: 2.0
    serial: <superuser required>
  Mobo: Micro-Star model: MPG X570 GAMING PLUS (MS-7C37) v: 2.0
    serial: <superuser required> UEFI: American Megatrends v: A.60
    date: 11/06/2019
CPU:
  Info: 6-core model: AMD Ryzen 5 3600 bits: 64 type: MT MCP cache: L2: 3 MiB
  Speed (MHz): avg: 2228 min/max: 2200/4208 cores: 1: 2200 2: 2200 3: 2199
    4: 2200 5: 2157 6: 2146 7: 2200 8: 2200 9: 2199 10: 2637 11: 2200 12: 2200
Graphics:
  Device-1: NVIDIA GA102 [GeForce RTX 3080 Lite Hash Rate] driver: nvidia
    v: 560.35.03
  Device-2: Logitech Logitech Webcam C925e driver: snd-usb-audio,uvcvideo
    type: USB
  Display: wayland server: X.Org v: 24.1.2 with: Xwayland v: 24.1.2
    compositor: gnome-shell driver: X: loaded: nvidia
    unloaded: fbdev,modesetting,nouveau,vesa gpu: nvidia,nvidia-nvswitch
    resolution: 1: 3440x1440~50Hz 2: 3440x1440~50Hz
  API: EGL v: 1.5 drivers: nvidia,swrast
    platforms: gbm,wayland,x11,surfaceless,device
  API: OpenGL v: 4.6.0 compat-v: 4.5 vendor: nvidia mesa v: 560.35.03
    renderer: NVIDIA GeForce RTX 3080/PCIe/SSE2
  API: Vulkan v: 1.3.290 drivers: N/A surfaces: xcb,xlib,wayland

### Printer

Bambu P1S

### How to reproduce

1. Download the linux app image from github (Bambu_Studio_ubuntu-v01.09.07.52-24.04_201d717a67193d236ffad5997305b33a.AppImage). This also affects Bambu_Studio_linux_ubuntu-v01.09.03.50_7f82a1f77720d730b7552ad20f10d99f.AppImage and BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage. The same error occurs with Orca Slicer.
2. Run the application e.g. from the command line to get debug txt.

### Actual results

The application starts and the homepage tab is fine, but in the Prepare and Preview tabs the sidebars and toolbars render, but the actual graphical content areas where the bed and models are rendered, nothing appears. The application is otherwise normal and toolbars/sidebars/menus can all be interacted with.

[image]

### Expected results

The bed and models render.

### Project file & Debug log uploads

Log file: [log.zip](https://github.com/user-attachments/files/17444288/log.zip)
Project file: N/A, problem occurs during app startup.

### Checklist of files to include

- [X] Log file
- [X] Project file

appositeit, 2024-10-19:
Sorry, I forgot to mention the critical point:

* Bambu Studio (and Orca Slicer) were working correctly under 24.04. I did a release upgrade to 24.10 and now Bambu Studio (and Orca Slicer) are both failing in an identical manner (as described above).

I have tried the following environment variables and they have had no effect:

GDK_SCALE=1
QT_QPA_PLATFORM=wayland
GDK_BACKEND=wayland
GDK_BACKEND=x11
WEBKIT_DISABLE_COMPOSITING_MODE=1

There's a bunch of GTK guff, when the application runs, but I believe it's all irrelevant:

(bambu-studio:3151280): Gtk-CRITICAL **: 18:43:13.022: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:3151280): Gtk-CRITICAL **: 18:43:13.022: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:3151280): Gtk-CRITICAL **: 18:43:13.022: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:3151280): Gtk-CRITICAL **: 18:43:13.023: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:3151280): Gtk-CRITICAL **: 18:43:13.023: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:3151280): Gtk-CRITICAL **: 18:43:13.023: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:3151280): Gtk-CRITICAL **: 18:43:13.024: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:3151280): Gtk-CRITICAL **: 18:43:13.024: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:3151280): Gtk-CRITICAL **: 18:43:13.024: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:3151280): Gtk-CRITICAL **: 18:43:13.025: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:3151280): Gtk-CRITICAL **: 18:43:13.025: gtk_widget_set_size_request: assertion 'width >= -1' failed

balliegojr, 2024-10-19:
I am facing the same issue. Broken after latest update.

appositeit, 2024-10-19:
If anyone else is stuck- a workaround is to use the docker container, which has it's own OS and isn't affected by this issue.

docker run -d   --name=bambustudio   --security-opt seccomp=unconfined `#optional`   -e PUID=1000   -e PGID=1000   -e TZ=Australia/Sydney   -e DARK_MODE=false `#optional`   -p 3000:3000   -p 3001:3001   -v /home/<user>/.config/BambuStudio:/config-v /home/<user>/Downloads:/downloads   --restart unless-stopped   lscr.io/linuxserver/bambustudio:latest

* Then go to: http://localhost:3000/ in your web browser.
* Log in to your Bambu account and it will sync all the settings for your printer.
* You can load files by importing them (File > Import  > Import 3MF...) from the /downloads folder which is mapped to the /home/<user>/Downloads/downloads folder.

Other than slightly awkward import, it's basically just normal Bambu. (Not optimal- I normally import by dragging the file into the window! But it totally works.)

MackBambu, 2024-10-24:
https://github.com/bambulab/BambuStudio/issues/4669
There are compatibility issues with the new NVIDIA driver, but the cause is still unclear for now.

appositeit, 2024-10-24:
Yes, #4669 looks right, thanks for the pointer. Much appreciated. I'll mark this as a duplicate.

appositeit, 2024-10-24:
Duplicate of #4669

closed 2024-10-24 by appositeit