bambulab/BambuStudio #5039: Slice doesn't genera supports

sfm67, 2024-10-19:
### Bambu Studio Version

1.9.7.5.2

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

Bambulab A1

### How to reproduce

trying to print this model with auto supports sets at 30°, no supports generated even if if rotate the model
https://makerworld.com/en/models/654983#profileId-581985

[image]

### Actual results

Print doesn't stick on plate

### Expected results

supports should be generated (i tryed values from 10 to 10°)

### Project file & Debug log uploads

[skeleton-dispenser-rotate.zip](https://github.com/user-attachments/files/17444475/skeleton-dispenser-rotate.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2024-10-21:
You painted support blockers. Remove them and the supports can be generated.

sfm67, 2024-10-21:
> You painted support blockers. Remove them and the supports can be generated.

no , just D/L from makerworld, open on bambu studio, slice and nothing else