bambulab/BambuStudio #5041: GUI can slice files that Command Line Cannot, citing "the print is empty"

vmorrisPolar, 2024-10-19:
### Bambu Studio Version

v01.09.07.52-20.04

### Where is the application from?

Bambu Lab github releases, Built from source by yourself or others

### OS version

Ubuntu

### Printer

Bambu X1C / (any printer)

### How to reproduce

Command: `./bambu-slicer --load-settings process.json --load-settings machine.json --slice 0 --debug 3 --arrange 0 --orient 0 --outputdir /slicing/ object.stl`

### Actual results

This is the resulting error from the slicer:
```
Initializing StaticPrintConfigs\n[2024-10-18 13:41:52.585078] [0x000079f121cc43c0] [warning] cli mode, Current BambuStudio Version 01.09.07.52
[2024-10-18 13:41:52.645298] [0x000079f121cc43c0] [warning] no filament colors found in projects\n[2024-10-18 13:41:52.664700]
[0x000079f121cc43c0] [error]   calc_exclude_triangles:Unable to create exclude triangles

[2024-10-18 13:41:52.676700] [0x000079f121cc43c0] [error]   plate 1: Nothing to be sliced, Either the print is empty or no object is fully inside the print volume before apply.\n\nrun found error, return -50, exit
```

### Expected results

[image]

It should slice exactly how it does in the app. For some reason, the headless version refuses to slice 50% of objects I'm throwing at it.

### Project file & Debug log uploads

[files and machine process json.zip](https://github.com/user-attachments/files/17445488/files.and.machine.process.json.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

vmorrisPolar, 2024-10-22:
Another issue, just trying to slice a simple cube. The STL, machine.json, filament.json, and process.json are all attached.
[bWg8OXNG-filament.json](https://github.com/user-attachments/files/17479892/bWg8OXNG-filament.json)
[bWg8OXNG-machine.json](https://github.com/user-attachments/files/17479893/bWg8OXNG-machine.json)
[bWg8OXNG-process.json](https://github.com/user-attachments/files/17479894/bWg8OXNG-process.json)
[bWg8OXNG-object.stl.zip](https://github.com/user-attachments/files/17479898/bWg8OXNG-object.stl.zip)

`Initializing StaticPrintConfigs\n[2024-10-22 17:09:22.089453] [0x0000734513de63c0] [warning] cli mode, Current BambuStudio Version 01.09.07.52\n[2024-10-22 17:09:22.123019] [0x0000734513de63c0] [warning] no filament colors found in projects\n[2024-10-22 17:09:22.137180] [0x0000734513de63c0] [error]   plate 1: Nothing to be sliced, Either the print is empty or no object is fully inside the print volume before apply.\n\nrun found error, return -50, exit...\n",`

lanewei120, 2024-10-28:
[error]   plate 1: Nothing to be sliced, Either the print is empty or no object is fully inside the print volume before apply
means no object on the plate
you can arrange this object before slice
use --arrange 1 instead of 0

vmorrisPolar, 2024-10-28:
Thanks for the reply. I have tried using `--arrange 1` but still get this error. Are you able to reproduce the issue via command line with the files I shared?

lanewei120, 2024-10-29:
[image]

vmorrisPolar, 2024-11-11:
We can mark this as closed. The issue seems to be intermittent. It won't slice, and then the exact same command and files will slice 8-10 minutes later.

closed 2024-11-11 by vmorrisPolar