bambulab/BambuStudio #5044: Model not sliced correctly with variable layer height and high quality profile, most layers missing in preview

Luro02, 2024-10-20:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab P1S 0.4 nozzle

### How to reproduce

1. Open the attached file
2. Slice the file

### Actual results

[image]

### Expected results

and this is how the model is supposed to look like

[image]

### Project file & Debug log uploads

[slicing-bug.zip](https://github.com/user-attachments/files/17449347/slicing-bug.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Luro02, 2024-10-20:
Hmmm, I think this is related to the profile not using arachne/"detect thin walls" not selected by default.

QingZhangBambu, 2024-10-21:
yes. but it's not suitable for a default setting for now