bambulab/BambuStudio #5047: In dark mode, the hover style is missing when selecting the print configuration.

Muggle-akko, 2024-10-20:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows10

### Printer

Bamboo A1 mini

### How to reproduce

1. Open Bamboo Studio; 2. Select the online model on the left sidebar; 3. Click on a model (it should come with multiple print configurations); 4. In the pop-up details box, click to select a print configuration in the print configuration column on the right

### Actual results

You can't clearly see which print configuration the user selected (that is, the hover style is missing, guessing that the background color may be the same because the hover color drinks the dark mode)

### Expected results

When using dark mode, the user should be able to confirm which print configuration he has selected.

### Project file & Debug log uploads

This is not a bug in the project

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Muggle-akko, 2024-10-20:
[image]
[image]

Muggle-akko, 2024-10-20:
[image]
[image]

Goldthing, 2024-10-20:
I came to report the same issue, only with MacOS version. It would be good if the light version also had better contrast, I have good eyesight but I could see this being difficult for anyone with impared vision.

realzorro, 2024-10-28:
Thank you for your feedback. Your suggestion is very good, and we have informed the development engineer to make the necessary changes. You should be able to see the optimization in the next few days.