bambulab/BambuStudio #5048: Painted "Brim Ears" are ignored when cloning objects

bzzzbzzzzz, 2024-10-20:
### Bambu Studio Version

v01.10.00.74

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 15 24A5309e

### Printer

Bambu Lab X1C

### How to reproduce

- [paint Brim Ears](https://github.com/bambulab/BambuStudio/issues/1393)
- select "painted" in Process -> Other -> Brim Type
- right click an object and select "Clone"

### Actual results

no painted brim ears on the cloned object

[image]

### Expected results

painted brim ears on the cloned object

### Project file & Debug log uploads

[box-opener_temp.3mf.zip](https://github.com/user-attachments/files/17450641/box-opener_temp.3mf.zip)
[Archive.zip](https://github.com/user-attachments/files/17450646/Archive.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MackBambu, 2024-10-21:
Confirmed the issue, thanks for reporting

MackBambu, 2024-10-29:
Fixed

closed 2024-10-29 by MackBambu