bambulab/BambuStudio #5049: AMS only shows Custom Filament from P1P machine profile even after P1S Enclosure upgrade

SchrodingersLad, 2024-10-20:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab P1P with Enclosure Upgrade

### How to reproduce

Click on 'Set Filaments to Use' in Prepare Tab
Create New Custom Filament
Create new filament named Test P1S PLA Pro using Copy Current Filament Preset with default Bambu Lab P1S 0.4 Nozzle on default Generic PLA Profile and Create Filament
[image]
Go to Device Tab and edit unknown filament in AMS Slot
[image]

### Actual results

Custom Filament Profile named Test P1S does not show up in list of filaments in AMS
[image]

### Expected results

Custom Filament should appear in list of filaments in AMS.
[image]

This result only happens if I create new filament (named Test P1P PLA Pro) using Copy Current Filament Preset with default Bambu Lab P1P 0.4 Nozzle on default Generic PLA Profile and Create Filament. (Note: only difference was I used the P1P machine profile instead of P1S machine profile)
[image]

I have a P1P that I upgraded to a P1S using the official enclosure upgrade kit. Despite this, the printer model still shows up as a P1P in the update section of the Device Tab in Bambu Studio. I believe this is causing my AMS to ignore Custom Filaments created under the P1S machine profile, and only show Custom Filaments created under and using the P1P machine profile. This is a huge oversight, seeing as my upgraded printer should function like a P1S and therefore use the P1S machine profile when printing.

### Project file & Debug log uploads

[Custom Filament Issue.zip](https://github.com/user-attachments/files/17450954/Custom.Filament.Issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Breaveheard, 2025-01-27:
Is still existing in Bambu Studio Version 1.10.1.50 and P1P Firmware 01.07.00.00

DanBao-Bambu, 2025-02-24:
The issue has been recorded and will be fixed as soon as possible.

boysie123, 2025-05-03:
you can link this to my one, this seems to be the root cause, i will test and check though;
https://github.com/bambulab/BambuStudio/issues/6579

@Breaveheard thx for finding that its been driving me nuts :)

Shame its all the way in the future now though and its still not fixed.