bambulab/BambuStudio #5052: Miscalculation on Support Material with multiple Filaments and Plates

FiireWiinter, 2024-10-20:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: AMD Ryzen 7 5800X
Memory: 32GB
Display Card: NVIDIA RTX 3070

### Printer

Bambu Lab P1P

### How to reproduce

1. Create a new project with 2 plates
2. Enable Supports (global or per object gives same result)
3. Add at least one object requiring supports to one of the plates
4. Add random object to other plate using ANOTHER FILAMENT (can also use supports, doesn't matter)
5. Slice both plates
6. Go to "All Plates Stats"
7. Observe total support filament

### Actual results

Total support filament used equals total model filament

### Expected results

Total Support Filament used should be the result of both filaments support

### Project file & Debug log uploads

[Sphere.zip](https://github.com/user-attachments/files/17452159/Sphere.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-10-21:
Please update the newest version ，and have a try. https://github.com/bambulab/BambuStudio/releases
[image]
The same issue:https://github.com/bambulab/BambuStudio/issues/4933

FiireWiinter, 2024-10-21:
Can confirm, newest version fixes this issue. Thanks for the quick reply

closed 2024-10-21 by FiireWiinter