bambulab/BambuStudio #5053: macos segfault

reedy, 2024-10-20:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.7

### Printer

P1S

### How to reproduce

Have BambuStudio open for an extended period, eventually, after sending to print, it'll segfault...

### Actual results

Segfaults

### Expected results

Not segfault

### Project file & Debug log uploads

[Archive.zip](https://github.com/user-attachments/files/17452323/Archive.zip)

[BambuStudio-2024-10-20-235111.ips.zip](https://github.com/user-attachments/files/17452324/BambuStudio-2024-10-20-235111.ips.zip)

[log-Archive.zip](https://github.com/user-attachments/files/17452327/log-Archive.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

reedy, 2024-10-22:
And again..

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [29253]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.09.07.52 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-10-22 02:27:12.3305 +0100
OS Version:            macOS 14.7 (23H124)
Report Version:        12
Anonymous UUID:        464EB762-6D2F-1499-C40A-5B51E69D4EF1

Time Awake Since Boot: 2600000 seconds

System Integrity Protection: enabled

Crashed Thread:        31

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x000000019ec52ae4

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [29253]

Application Specific Information:
abort() called

Thread 0:: bambustu_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x19ed06b8c __ulock_wait + 8
1   libsystem_pthread.dylib       	       0x19ed4848c _pthread_join + 608
2   BambuStudio                   	       0x103e315f8 boost::thread::join_noexcept() + 188
3   BambuStudio                   	       0x103a9db08 Slic3r::GUI::Job::join(int) + 116
4   BambuStudio                   	       0x103b86170 Slic3r::GUI::SelectMachineDialog::prepare_mode(bool) + 384
5   BambuStudio                   	       0x103b8b94c Slic3r::GUI::SelectMachineDialog::show_status(Slic3r::GUI::PrintDialogStatus, std::__1::vector<wxString, std::__1::allocator<wxString>>) + 7836
6   BambuStudio                   	       0x103b9923c Slic3r::GUI::SelectMachineDialog::Show(bool) + 48
7   BambuStudio                   	       0x1058fcd98 wxDialog::EndModal(int) + 68
8   BambuStudio                   	       0x10376e35c Slic3r::GUI::Plater::print_job_finished(wxCommandEvent&) + 140
9   BambuStudio                   	       0x105a3da64 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
10  BambuStudio                   	       0x105a3d744 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
11  BambuStudio                   	       0x105a3d568 wxEvtHandler::ProcessEvent(wxEvent&) + 56
12  BambuStudio                   	       0x105a3d3b8 wxEvtHandler::ProcessPendingEvents() + 236
13  BambuStudio                   	       0x1059a1760 wxAppConsoleBase::ProcessPendingEvents() + 112
14  BambuStudio                   	       0x105a46274 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 60
15  CoreFoundation                	       0x19ee2487c __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
16  CoreFoundation                	       0x19ee24768 __CFRunLoopDoObservers + 536
17  CoreFoundation                	       0x19ee23d94 __CFRunLoopRun + 776
18  CoreFoundation                	       0x19ee23434 CFRunLoopRunSpecific + 608
19  HIToolbox                     	       0x1a95ce19c RunCurrentEventLoopInMode + 292
20  HIToolbox                     	       0x1a95cde54 ReceiveNextEventCommon + 260
21  HIToolbox                     	       0x1a95cdd30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
22  AppKit                        	       0x1a2682cc8 _DPSNextEvent + 660
23  AppKit                        	       0x1a2e794d0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
24  BambuStudio                   	       0x10596daa0 wxGUIEventLoop::DoDispatchTimeout(unsigned long) + 180
25  BambuStudio                   	       0x105a4677c wxCFEventLoop::DispatchTimeout(unsigned long) + 44
26  BambuStudio                   	       0x105a4689c wxCFEventLoop::OSXDoRun() + 172
27  BambuStudio                   	       0x10596df70 wxModalEventLoop::OSXDoRun() + 88
28  BambuStudio                   	       0x105a46958 wxCFEventLoop::DoRun() + 40
29  BambuStudio                   	       0x1059c6004 wxEventLoopBase::Run() + 132
30  BambuStudio                   	       0x1058fcc74 wxDialog::ShowModal() + 100
31  BambuStudio                   	       0x103710790 Slic3r::GUI::DPIAware<wxDialog>::ShowModal() + 124
32  BambuStudio                   	       0x103724980 Slic3r::GUI::Plater::priv::on_action_print_plate(Slic3r::GUI::SimpleEvent&) + 280
33  BambuStudio                   	       0x105a3da64 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
[... 663 lines cut ...]
      "experimentId" : "6643969b3099cf28e049862f",
      "deploymentId" : 400000007
    }
  ]
}
}

Model: Mac14,3, BootROM 11881.1.1, proc 8:4:4 processors, 24 GB, SMC 
Graphics: Apple M2, Apple M2, Built-In
Display: DELL S2721DGF, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4388), wl0: Apr  4 2024 20:56:02 version 23.30.58.0.41.51.138 FWID 01-60807d6e
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: Back-UPS RS 550G FW:857.L4 .I USB FW:L4
USB Device: USB2.1 Hub
USB Device: USB3.1 Hub
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.
```

walterwongbbl, 2024-10-28:
@reedy 
We have attempted to fix this issue and will release a new version in the future.