bambulab/BambuStudio #5058: Change "type" on multiple objects

Rupp3r, 2024-10-21:
**Is your feature request related to a problem? Please describe.**
I want to make multiple parts negatives. In this case multiple text objects making up the whole text.
Selecting multiple objects in the "Objects" tab removes this option from the right click menu.

**Describe the solution you'd like**
The possibility to change "Type" on multiple objects at a time.

**Describe alternatives you've considered**
Doing it manually for the 25 objects by selecting each one - right clicking - selecting "Change type" - setting type to "Negative Part" - Ok.

**Additional context**
Screenshot when right clicking a single object:
[image]
Screenshot when right clicking multiple objects:
[image]

Bajerok, 2025-05-10:
Did they add this?

vanowm, 2025-06-08:
Duplicate of [#1400](https://github.com/bambulab/BambuStudio/issues/1400)

Highspeedfutzi, 2026-02-26:
This is fixed now. You can select multiple objects and change the type.

Rupp3r, 2026-02-26:
Can confirm that it is fixed in #9055.... Anyone know why the issues have not been closed? 😅