bambulab/BambuStudio #5059: Auto Tree Support Prints in Midair

dragazo, 2024-10-21:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Mint 20 (Ubuntu 20.04)

### Printer

Bambu Lab A1

### How to reproduce

1. Open the attached project file.
1. All settings are set to reproduce the issue, so just click slice plate.
1. Observe the low support layers (layers 1-10) near the center area of the main crab body.

### Actual results

After building up the initial layer on the ground, the slicer proceeds to build up a few layers of the auto tree supports around the perimeter. However, at around layer 7 or so, it begins to split the main auto tree cluster in several sub-trees, but in this case does so by printing supports in midair: [image].

I did not notice this at first and attempted the print, but it failed in exactly this location for presumably this exact reason. So I know at least that it is not simply a visualizer issue in the preview.

### Expected results

If the tree cluster needs to be split into sub-trees, it should start building up their walls from the very beginning rather than waiting until the fork in their surface topologies (which I believe is what it actually is doing).

### Project file & Debug log uploads

[project.zip](https://github.com/user-attachments/files/17461797/project.zip)
[log.zip](https://github.com/user-attachments/files/17461822/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2024-10-22:
Yes, that's one of the known issues with organic tree support (the current default). You can try hybrid support instead.

dragazo, 2024-10-24:
When using the other tree types, I often get clipping issues where non-leaf segments adhere to the model.

But since this mid-air support problem is a know issue, I'll close this as a duplicate. Thanks!

closed 2024-10-24 by dragazo