bambulab/BambuStudio #5064: Aft Right Purge Line Missing

mltriebe, 2024-10-23:
### Bambu Studio Version

v01.10.00.74

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

Slice an object with the .4 nozzle selected and there is no purge line.  The purge line is there for the .2 and .6 profiles.

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[Uploading thread (v4).zip…]()

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-10-23:
[image]
there is no problem~

Haidiye00, 2024-10-23:
Hello, can you record a GIF? Thank you

ttt888zzz, 2024-10-23:
Thank you for your feedback.It's not a bug. 
This is the wiper print line defined in start gcode, and the position of this line in start gcode has changed, and it's in line with the calibration line.

closed 2024-10-23 by ttt888zzz

mltriebe, 2024-10-23:
Only on the .4 nozzle profile?  That seems odd.

mltriebe, 2024-10-23:
> ![image](https://private-user-images.githubusercontent.com/132528822/379067990-4f3a8e5e-689f-4ef3-87a1-57f4243cd2af.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

It works with the .2 nozzle you have selected.  Try the .4 nozzle.

ttt888zzz, 2024-10-23:
[image]
The imge and 3mf file cannot be downloaded，Could you pls upload the 3mf file  and the img with the  .4 nozzle? thank you

mltriebe, 2024-10-23:
[thread (v4).zip](https://github.com/user-attachments/files/17484663/thread.v4.zip)
[image]

The zip file should be the 3mf.

MartinChan3, 2024-10-30:
[1.10 beta2](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81) has solved this problem.

danielo515, 2024-11-21:
I was not sure what the technical term was for this, and I found nothing in the last release notes. Too bad you guys decided something this obvious is not worth mentioning.
To be honest, this line was annoying to remove and felt redundant, glad you removed it, but it was worth mentioning for sure