bambulab/BambuStudio #5066: Too small Scarf start height (Cannot set Scarf height to 0%)

silvernull, 2024-10-23:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

A1 Mini

### How to reproduce

Go to Prepare > Seam > enable Scarf joint seam (experimental) > Scarf start height

attempt to set a value smaller than 50% (you can set a value in mm, but not 0
[image]

### Actual results

[image]
[image]

### Expected results

Either the pop-up needs adjusting or the scarf height should be possible to set to 0%.

### Project file & Debug log uploads

[scarf height issue.zip](https://github.com/user-attachments/files/17489649/scarf.height.issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-10-24:
@silvernull can you try the latest version? it allows to use 0

ttt888zzz, 2024-10-25:
Please visit the website at https://github.com/bambulab/BambuStudio/releases and try again

closed 2025-05-23 by QingZhangBambu