bambulab/BambuStudio #5068: Tolerance issue

amw123456789, 2024-10-23:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

A1

### How to reproduce

I can print the same item on my A1 mini and it does just fine but when I print it on the A1 the tolerance is off by about .20mm in an open circle spot of the print. I did recently add the AMS lite to my A1 could that make a difference? 

### Actual results

N/A

### Expected results

N/A

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-10-24:
HI [amw123456789](https://github.com/amw123456789): Could you pls provide the source 3mf file to help check the issue，thank you

amw123456789, 2024-10-24:
[150% candy dispenser project-test.zip](https://github.com/user-attachments/files/17504853/150.candy.dispenser.project-test.zip)

amw123456789, 2024-10-24:
[image]
It is this dimension that becomes different during printing on the A1 with the AMS lite. I did recently just purchase the AMS and added it to my printer. I also used the official top mount that i printed from makerworld. When i get time i plan i removing the AMS and seeing if that makes a difference.