bambulab/BambuStudio #5070: "Failed to publish login event" when trying to connect to printer by IP

therealmrfox, 2024-10-24:
### Bambu Studio Version

linux_fedora-v01.10.00.74

### Where is the application from?

Bambu Lab github releases

### OS version

Debian 6.9.10-1rodete4 (2024-08-02) x86_64 GNU/Linux

### Printer

P1S

### How to reproduce

1. Open Bambu Studio 
2. Under " Device " > " No Printer + " > " + Bind With Access Code", enter IP and access code of printer 

### Actual results

Immediately shows "Failed to publish login request" (or, occasionally, crashes).

Per the [troubleshooting guide](https://wiki.bambulab.com/en/bambu-studio/troubleshoot/login-binding-issue) this indicates a problem connecting with the Bambu cloud, which makes no sense given I'm attempting to *locally* connect.

Printer works perfectly fine once discovered via SSDP, so there is no connection issue at play here. 

[image]

### Expected results

Connect to the printer.

### Project file & Debug log uploads

Log file seems to contain usernames / user IDs, so, I will not publicly share it. But here are some bits that look relevant:

    2024-10-23 21:14:46.665595[Thread 0x00007f532dc006c0]:bind_detect error: network_agent=0x1bdd2040, ret=-2, dev_ip=192.168.6.119
    2024-10-23 21:14:48.334476[Thread 0x00007f52e7a006c0]:bind_detect error: network_agent=0x1bdd2040, ret=-2, dev_ip=192.168.6.119
    2024-10-23 21:14:49.643542[Thread 0x00007f5352c006c0]:bind_detect error: network_agent=0x1bdd2040, ret=-2, dev_ip=192.168.6.119

### Checklist of files to include

- [ ] Log file
- [ ] Project file

WisdomCode, 2024-11-16:
Got the same issue here with 1.10.0.89 in flatpak with an A1.

blackfyre, 2024-11-18:
Got the same issue here with flatpak (ubuntu 22.04, v01.10.01.50) with a P1P (1.06.01.02).

DeCodeerHeer, 2024-11-18:
Also here same issue with FlatPak on Ubuntu 24.04.1 with P1S

blackfyre, 2024-11-19:
My guess that the issues is in the network plugin...

walterwongbbl, 2024-12-02:
Hi. @therealmrfox  
What would happen if you tried ping your printer.  Is there a network connection between them

therealmrfox, 2024-12-02:
Hi @walterwongbbl ,

Yes, there is a network connection. Everything works fine once once the printer is discovered via SSDP.

azbulutlu, 2024-12-15:
+1 to this with the v01.10.01.50 appImage (ubuntu 24.04 and P1S)

Yakusari, 2024-12-27:
+1 for Version 1.10.1.50 on Ubuntu 24.04 with the A1

Teonyr, 2025-01-03:
+1 still a Problem in v. 1.10.1.50 on Fedora Flatpak

dcdevotion, 2025-01-03:
+1, same here in 1.10.1.50 on Ubuntu 24.04

tabre, 2025-01-05:
Same issue here on Arch

righteousxkarma, 2025-01-09:
+1 same. Mint 22 (ubuntu 24.04) with the Ubuntu AppImage, via LAN. I can ping it no problem. 

NAME="Linux Mint"
VERSION="22 (Wilma)"
ID=linuxmint
ID_LIKE="ubuntu debian"
PRETTY_NAME="Linux Mint 22"
UBUNTU_CODENAME=noble

recursiveGecko, 2025-01-09:
Same issue, discussed in depth in ticket US241218736001 which contains logs, screenshots, video and a Wireshark trace showing the discrepancy:

1. When connecting to the printer through auto-discovery, Bambu Studio successfully connects to the printer on port **8883**

2. When connecting to the printer by IP, Bambu Studio attempts to connect to the printer on port **3000** and receives a TCP reset, indicating that it's not connecting to the right port.

I installed Bambu Studio through flathub.

DaveDavenport, 2025-01-11:
If I redirect port 3000 to port 8883 in my router, it indeed connects to the printer just fine using the access code method.

Yethal, 2025-01-16:
Fixed in newest appimage for me

KentBrockman, 2025-01-17:
I'm also having this issue on P1S in LAN only mode.

Firmware: 01.07.00.00

[1.10.1.50](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50) and [1.10.2 Public Beta](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.02.64) produce the same result.

I can ping the printer and connect to FTPS from the host running BambuStudio. Please advise.

therealmrfox, 2025-01-17:
Still broken in Bambu_Studio_linux_fedora-v01.10.02.64.AppImage and Bambu_Studio_ubuntu-24.04_PR-5731.AppImage.

C'mon Bambu, this should be an easy fix... Based on [this comment](https://github.com/bambulab/BambuStudio/issues/5070#issuecomment-2579859209) perhaps as easy as fixing a port number someone fat-fingered?

Bobylein, 2025-01-19:
Same problem with the 1.10.1 public release on Flathub with a Manjaro Linux system

kayvax, 2025-01-19:
Broken in Bambu Studio 1.10.1.50

JacobRaphael, 2025-01-19:
I fixed the problem. I wanted to switch from cloud to local and enable LAN mode. I tried adding the printer again but received an error. However, the printer was still listed as an 'Other Device' below the 'Bind with Access Mode' section. I then entered only the access code, and it worked.

recursiveGecko, 2025-01-19:
@JacobRaphael That works but only in certain scenarios. You're not connecting to the printer "by IP" (what this issue is about), but instead through the auto-discovery in Bambu Studio which only works if you're on the same network subnet.

btittelbach, 2025-01-19:
confirm I could add it in LAN mode 3 months ago afair. Then switched to cloud and wanted to switch back to LAN mode today, but no luck. Same error message. IP and access code are correct. Printer is pingable and reachable.

therealmrfox, 2025-01-20:
Given recent announcements, it's probably safe to say Bambu has completely lost interest in making their products function in local mode. Instead they are pushing users towards their cloud (and, I'm sure, requiring a paid subscription isn't far behind) -- security, reliability, common sense, respect for their customers, and functionality that was promised when we bought the printer all be damned. 

@walterwongbbl this is shameful. We are asking for basic functionality, instead we get enshittification. 

As for me, I have added a firewall rule to prevent my P1S from ever talking to the internet, and will stop using Bambu Studio. I don't trust Bambu not to force-install a firmware update without my consent that completely locks me out of using the printer locally.

blackfyre, 2025-01-20:
I believe that it's time to look into getting Klipper on the printers/toolheads/AMS/...

btittelbach, 2025-01-21:
issue persists .... afair this worked 3 months ago, but not anymore when re-adding printer with IP and correct access-code. Printer is still in different subnet but same LAN

btittelbach, 2025-01-21:
I managed to get a step further, by using smcroute to forward the printers SSDP packets to my PC subnet ..

now it goes to the motions of accepting the Access Code, closes the dialog-box and informs me the printer will appear in the list shorty...   Though it never does.

drerik, 2025-01-22:
Check if you have a firewall enabled on your client machine.
I have firewalld enabled by default and it seems that bambu studio is listening on UDP port 2021:
```bash
$ sudo netstat -lnp | grep 2021
udp        0      0 0.0.0.0:2021            0.0.0.0:*                           31913/bambu-studio 
```

As i do not like the "just disable your firewall" "solution", I allowed the udp port when I am on my home network:
```bash
sudo firewall-cmd --permanent --zone=home --add-port=2021/udp
```

I tested several times, and it seems to be a stable fix.

recursiveGecko, 2025-01-24:
As a workaround, running this script to generate spoofed SSDP announcements works for me. My printer is on a different subnet and I can connect to it, watch the video feed, and start print jobs.

https://github.com/jonans/bsnotify

dentex, 2025-02-14:
Same issue this evening on a A1 in LAN only mode, with latest version 01.04.00.00. As far as I understand, this is an error regarding the connection to the bambulab cloud... that should not be relevant in LAN only mode. For reference, see https://wiki.bambulab.com/en/bambu-studio/troubleshoot/login-binding-issue 

[image]

I switched to Orca Slicer, and connection is OK... I finally started printing.

lanrat, 2025-02-14:
When I get this error in LAN mode, usually restarting both the printer and Bambu Studio fixes the issue.
Its not a solution, but a workaround for the time being.

dentex, 2025-02-14:
> When I get this error in LAN mode, usually restarting both the printer and Bambu Studio fixes the issue. Its not a solution, but a workaround for the time being.

Thanks, next time I'll try. I did restart just the slicer, but not the printer. Instead I recycled a new connection password and disable/re-enable the wi-fi connection.

dentex, 2025-02-20:
Various reboots and powering cycles, but this is still a thing.

jburnash, 2025-02-26:
Still happening to me. I tried basically everything posted here, with no success.

Bambu P1s on my local network. Bambu Studio Flatpak running on Fedora 40

Worked absolutely perfectly for over a month until the day before yesterday. I had made no changes to any application configs, my network.

I *had* patched my workstation Tuesday morning - which suggests firewall to me. Turned firewalld off completely - still get "Failed to publish login request"

Infuriating

jburnash, 2025-03-02:
Update: FWIW, for me, the answer was simply logging out of my printer from the front control panel, then logging in again.

Maddening, but just in case it's useful to anyone else.

dentex, 2025-03-02:
There is no login here at play, since what's not working is the discovery of the printer in LAN only mode via IP address and acces code.

ianfabs, 2025-03-23:
> Check if you have a firewall enabled on your client machine. I have firewalld enabled by default and it seems that bambu studio is listening on UDP port 2021:
> 
> $ sudo netstat -lnp | grep 2021
> udp        0      0 0.0.0.0:2021            0.0.0.0:*                           31913/bambu-studio 
> 
> As i do not like the "just disable your firewall" "solution", I allowed the udp port when I am on my home network:
> 
> sudo firewall-cmd --permanent --zone=home --add-port=2021/udp
> 
> I tested several times, and it seems to be a stable fix.

Ran into this issue on Ubuntu 20ish and the above was the solution that worked for me (thank you @drerik !)

Specifically, it was this:
```sh
sudo ufw allow 2021/udp
```

albfan, 2025-03-30:
still not much specific, but after disable firewall
```
#avoid restart
sudo systemctl disable firewalld.service 
#stop firewall
sudo systemctl stop firewalld.service 
```

and start bambu studio (appimage) with sudo

```
sudo ~/.local/bin/bambu-studio
```

bambulab studio is able by itsef to discover the printer, asking only for access code, and success to connect

using ip and access code still fails the same

blaggacao, 2025-04-01:
> > Check if you have a firewall enabled on your client machine. I have firewalld enabled by default and it seems that bambu studio is listening on UDP port 2021:
> > $ sudo netstat -lnp | grep 2021
> > udp        0      0 0.0.0.0:2021            0.0.0.0:*                           31913/bambu-studio
> > As i do not like the "just disable your firewall" "solution", I allowed the udp port when I am on my home network:
> > sudo firewall-cmd --permanent --zone=home --add-port=2021/udp
> > I tested several times, and it seems to be a stable fix.
> 
> Ran into this issue on Ubuntu 20ish and the above was the solution that worked for me (thank you [@drerik](https://github.com/drerik) !)
> 
> Specifically, it was this:
> 
> sudo ufw allow 2021/udp

✅ Worked, ty

ahoffer, 2025-04-12:
_Connect the printer using IP and access code._

Error message is "Failed to publish login request."

- Ubuntu 24, up-to-date with Canonical's package
- No firewall
- Running **OrcaSlicer_Linux_AppImage_Ubuntu2404_V2.3.0.AppImage**
- Running is as normal user
- I can ping the printer's IP
- The access code is correct

devcomputer, 2025-09-16:
bambulabs is pathetic.
works for me on mint22,debian24 but not on cachyOS using the exact same AppImage.

unitrix0, 2025-12-10:
For me it was the firewall blocking the traffic.

fuggla, 2026-01-04:
Allowing 2021/udp incoming worked for me.

```bash
# firewalld / fedora
sudo firewall-cmd --add-port 2021/udp --permanent
sudo firewall-cmd --reload

# ufw / ubuntu
sudo ufw allow 2021/udp
```