bambulab/BambuStudio #5080: Bambu Slicer Removing small overhangs

wxyzsupermod, 2024-10-26:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

AMD Ryzen 7950x3d
Nvidia RTX 4090
32GB DDR5

### Printer

Bambu Lab P1s

### How to reproduce

Open .step file with 40 degree overhang with 1mm ridges on print 
[image]
Slice in Classic Wall Generator(Default)
[image]
OverHangs are removed
Slice in Arachne Wall Generator
[image]
Overhangs are retained

### Actual results

Slicing with Classic Wall Generator Results in the overhangs being removed 
[image]
Slicing with Arachne Wall Generator Restults in the overhangs being retained
[image]

### Expected results

Slicing in the default classic wall generator should retain the overhangs in the model

### Project file & Debug log uploads

[40 Deg Front Cam Mount M2 R1 TPU C.zip](https://github.com/user-attachments/files/17528444/40.Deg.Front.Cam.Mount.M2.R1.TPU.C.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-11-12:
for classic mode, this part is too small to generate wall.
[image]

QingZhangBambu, 2025-05-23:
close for no response

closed 2025-05-23 by QingZhangBambu