bambulab/BambuStudio #5081: Slicer Error

tgreutmann, 2024-10-26:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU Ryzen 9 32GByte Memory NVIDIA GeForce RTX 3060 Ti

### Printer

Bambu Lab X1C

### How to reproduce

1. Go to Plate #3 in the project
2. Slice the object Hotas Support (4).stl

### Actual results

Around layers 250 to 260, The slicer creates an odd band outside the STL objects, one filament thread wide and several layers high, cross-connecting two edges across the object. See also screenshots attached, one showing the slicer result, the other showing the camera snapshot during printing. I have marked the band with arrows. No logfile was produced.

### Expected results

No such band should occur.

### Project file & Debug log uploads

[Downloads.zip](https://github.com/user-attachments/files/17529789/Downloads.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tgreutmann, 2024-10-26:
Two additional observations:

- After printing it looks like the banding layers are actually a misplaced outermost wall layer of the adjacent walls. Somehow the slicer makes creates them in straight line between the edges, instead of placing them in an L-shape.
- I turned the object by 90 degrees around the Z-axis and then ran the slicer again. The strange banding disappeared. So I think that it is a rare object constellation where the slicing algorithm is thrown off the proper course.

ttt888zzz, 2024-10-28:
Please update the newest version，and have a try.https://github.com/bambulab/BambuStudio/releases

QingZhangBambu, 2025-05-23:
works fine on latest version, it had been fixed. close

[image]

[image]

closed 2025-05-23 by QingZhangBambu