bambulab/BambuStudio #5083: SegFault on Linux when starting the Camera

einsz, 2024-10-26:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Built from source by yourself or others

### OS version

Arch Linux

### Additional system information

Bambu Studio installed from AUR
aur/bambustudio-bin

### Printer

P1S

### How to reproduce

Open Bambu Studio, go to Device, press play on camera

### Actual results

Bambu Studio crashes with a segfault

### Expected results

Camera stream should open

### Project file & Debug log uploads

running Bambu Studio in gdb gives this output at the time of crash:

```
(...)
[Thread 0x7fff44c006c0 (LWP 17974) exited]
[New Thread 0x7fff44c006c0 (LWP 18195)]
[New Thread 0x7ffef2a006c0 (LWP 18196)]
[Thread 0x7ffef2a006c0 (LWP 18196) exited]
[New Thread 0x7ffef2a006c0 (LWP 18197)]
[New Thread 0x7ffef16006c0 (LWP 18198)]
[New Thread 0x7ffef0c006c0 (LWP 18200)]

Thread 27 "source:src" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff516006c0 (LWP 17959)]
0x00007fffee9337d6 in ?? () from /usr/lib/libc.so.6
```

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-10-28:
Please try version 1.10 beta.

einsz, 2024-10-28:
Camera works without segfault in 1.10 beta

araneusdiadema, 2025-10-31:
The issue is back or still there. On arch linux, latest version 2.3.0 crashes on device tab when you try to open camera. It happens with appimage, aur/bambustudio-bin and I also tested flatpak and it crashes too.