bambulab/BambuStudio #5084: Lost modifications

contactzero, 2024-10-26:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS Sonoma 14.6.1

### Printer

A1 Mini

### How to reproduce

SETUP
Open any model from the website.
Modify it anyway you like; eg. Add plate, copy paste part on new plate, add negative part.

ERREUR
Press (Mac) Command-N.

PROBLEM
All modifications lost.

DESCRIPTION
Maybe you wanted to HIDE the window (Command-M) and hit the wrong key.
Maybe you want to experiment in a new document.

It what universe of computer applications does creating a new document destroy the previous document?! Without even asking to save the current document?

BUG. BUG. and poor UI / user experience.

### Actual results

DATA LOSS

### Expected results

NEW WINDOW

### Project file & Debug log uploads

[any file.3mf.zip](https://github.com/user-attachments/files/17530722/any.file.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

contactzero, 2024-10-30:
> image.png (view on web)
> Will there be a pop-up reminder when pressing Ctrl+N. Don't you have it there

Hello and thank you for your response,Strangely, no such message, thus the reason for my inquiry. Seeing that you have makes more sense that it is a unique problem on my machine.I have several computers, I will try on others computers and get back with you.Kind regards,Geoff LacyEnvoyé de mon iPhone

Haidiye00, 2024-10-31:
you can clear you old choice
[image]
[image]

closed 2024-10-31 by Haidiye00

JohnSmithCYM, 2024-12-13:
@contactzero Have you tried @Haidiye00 's method to solve your problem?Did that work?

Haidiye00, 2024-12-13:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you