bambulab/BambuStudio #5092: 自动朝向完全失效，只能手动修改

muyir123, 2024-10-28:
### Bambu Studio Version

1.10.0.81

### Where is the application from?

Bambu Lab github releases

### OS version

win11

### Printer

A1C

### How to reproduce

自动朝向点击后完全没反应

### Actual results

自动朝向点击后完全没反应

### Expected results

自动朝向

### Project file & Debug log uploads

无需日志，可直接复现

### Checklist of files to include

- [ ] Log file
- [ ] Project file

fahaaaxiki, 2024-10-29:
您好，我在我的电脑上使用自动朝向功能是正常生效的，您是否可以录制一个您操作时失效的视频以及将对应的3mf文件提供给我们？非常感谢。
[image]

muyir123, 2024-10-29:
好的，这个压缩包内包含视频和3MF文件，这次测试的发现可能是自动朝向算法有问题，没有正确朝向，我手动摆放好后点自动朝向，他又自动竖起来了
[无法自动朝向 - BambuStudio 2024-10-29 16-07-57.zip](https://github.com/user-attachments/files/17553680/-.BambuStudio.2024-10-29.16-07-57.zip)
@fahaaaxiki @Haidiye00

muyir123, 2024-10-29:
> 您好，我在我的电脑上使用自动朝向功能是正常生效的，您是否可以录制一个您操作时失效的视频以及将对应的3mf文件提供给我们？非常感谢。
已上传，但是视频中无法录制到版本号，版本号为1.10.0.81，目前最新的测试板

tangketan, 2024-10-29:
目前使用的自动朝向算法比较原始，确实存在一些朝向错误的场景。完全修复它很困难。目前只能靠用户手动朝向。

muyir123, 2024-10-29:
> 目前使用的自动朝向算法比较原始，确实存在一些朝向错误的场景。完全修复它很困难。目前只能靠用户手动朝向。

正式版本这个模型自动朝向没问题，测试版本出现的，您可以导出模型后在正式版本试一下

tangketan, 2024-10-30:
@muyir123 感谢你的信息！确实是这样。我们将立刻修复这个问题。

muyir123, 2024-10-30:
> @muyir123 感谢你的信息！确实是这样。我们将立刻修复这个问题。

@tangketan 好的，麻烦您了