bambulab/BambuStudio #5110: Fuzzy skin with modifier

hottr, 2024-10-30:
### Bambu Studio Version

1.9.7

### Where is the application from?

Bambu Lab github releases

### OS version

win 10

### Additional system information

ryzen 9 5950x

### Printer

P1S

### How to reproduce

Slice model and check the edge of section with fuzzy skin. compare to orca slicer

### Actual results

Buldging on the edge of section with fuzzy skin

### Expected results

no buldging like in Orca slicer

### Project file & Debug log uploads

[Fuzzy skin.zip](https://github.com/user-attachments/files/17572754/Fuzzy.skin.zip)
project file and results in Bambu Studio and Orca slicer

### Checklist of files to include

- [ ] Log file
- [X] Project file

ttt888zzz, 2024-10-31:
Please update the newest version，https://github.com/bambulab/BambuStudio/releases

hottr, 2024-11-02:
Thanks, 
Works as expected in 1.10.00.81

closed 2024-11-04 by QingZhangBambu