bambulab/BambuStudio #5118: Layer reported by Printer and Slicer Device screen, differs to Preview in Slicer

WanaGo, 2024-11-01:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

N/A I think, however
Ryzen 9 7900X, 64GB RAM, RTX3080 GPU, NVME SSD etc

### Printer

X1C

### How to reproduce

1. Slice an item
2. Send to printer
3. Look at printer status showing what layer its up to, and also Device screen of the slicer
4. Compare to the Preview screen in the Slicer. The total number of layers are not the same.

Discovered by looking at the preview screen and trying to work out where I was on the print by sliding the right hand vertical slider up and down, to see when something was about to be printed, and found the numbers didnt relate at all

### Actual results

Example 1: 692 on the Printer/Device screen, vs 1076 on the Preview screen.
[image]
[image]

Example 2: 644 on the Printer/Device screen, vs 1007 on the Preview screen.
[image]
[image]

### Expected results

The Device Screen, Printer, and total layers on the Preview screen, should all surely match the total number of layers in the print. I dont understand how they can not relate.

### Project file & Debug log uploads

[Project.zip](https://github.com/user-attachments/files/17595915/Project.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-11-04:
the latest version is fine. please try it.
[image]

ttt888zzz, 2024-11-12:
Please update the newest version，https://github.com/bambulab/BambuStudio/releases

closed 2024-11-12 by WanaGo