bambulab/BambuStudio #5119: slicer crashes with larger models

HPVA, 2024-11-01:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

	
Processor	Intel(R) Xeon(R) W-10855M CPU @ 2.80GHz   2.81 GHz
Installed RAM	32.0 GB (31.8 GB usable)

Product ID	00391-50000-00000-AAOEM
System type	64-bit operating system, x64-based processor

### Printer

p1s

### How to reproduce

select a model from the online models within the slicer, such as a moon lithograph, slice the model, bambu studio crashes

### Actual results

bambu studio simly closes, no error dialog displayed

### Expected results

model should slice and allow it to be printed

### Project file & Debug log uploads

[debug_network_Thu_Oct_31_21_04_02.log.zip](https://github.com/user-attachments/files/17595964/debug_network_Thu_Oct_31_21_04_02.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-11-01:
HI [HPVA](https://github.com/HPVA)
1.Please try the following steps，https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing
2. If it doesn't work，could you pls provide the 3mf file？thank you.

HPVA, 2024-11-01:
Thank you for the reply. Unfortunately this did not resolve the issue. Find
attached the 3mf file. The problem does however occur with any large or
complicated model
 %50 moon.zip
<https://drive.google.com/file/d/15Fu9hW0R1Ly9kMGw0o6dBV6rueuXgfoD/view?usp=drive_web>
Thank you