bambulab/BambuStudio #5126: Black screen on prepare tab

awalk94, 2024-11-01:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 

### Additional system information

CPU - AMD Ryzen 7 3700X 8-Core Processor - 8 Cores
RAM - 32 GB
AMD Radeon RX 5700 XT

### Printer

Bambulab P1S

### How to reproduce

open bambu studios and the prepare tab shows a black screen where object should be, has worked fine since ive had it till now. everything else works fine . Have reinstalled program twice
[image]

### Actual results

[image]

### Expected results

should not be black an able to see object and editing buttons

### Project file & Debug log uploads

its any file.Graphics

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Roughrider, 2024-11-01:
I have the exact same thing going on. Was fine this morning.

unhookeddotnet, 2024-11-02:
likewise, this just started today. 
Ryzen 7 7735HS with Radeon Graphics
680M
it's happening in orca 220 as well. 
windows update?

unhookeddotnet, 2024-11-02:
amd software app... image sharpening for some reason seems to trigger this. 
I just turned everything off on the default profile and it reverted back to normal.
odd because neither drivers or bambu had updated and everything was working fine after windows updated.

awalk94, 2024-11-02:
Legend thanks unhookeddotnet that done the trick

Haidiye00, 2024-11-07:
> AMD 软件应用程序...由于某种原因，图像锐化似乎会触发这种情况。我刚刚关闭了默认配置文件上的所有内容，它就恢复正常了。奇怪的是，驱动程序或 Bambu 都没有更新，并且在 Windows 更新后一切正常。

good idea

guanyun-gudujian, 2024-12-27:
> ### Bambu Studio Version
> 1.9.7.52
> 
> ### Where is the application from?
> Bambu Lab Official website
> 
> ### OS version
> Windows 11
> 
> ### Additional system information
> CPU - AMD Ryzen 7 3700X 8-Core Processor - 8 Cores RAM - 32 GB AMD Radeon RX 5700 XT
> 
> ### Printer
> Bambulab P1S
> 
> ### How to reproduce
> open bambu studios and the prepare tab shows a black screen where object should be, has worked fine since ive had it till now. everything else works fine . Have reinstalled program twice ![bambu lab screenshot](https://private-user-images.githubusercontent.com/187000101/382412900-3a97f448-c5fa-44ce-8651-ff4c5b2ea75e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUyODI0MDgsIm5iZiI6MTczNTI4MjEwOCwicGF0aCI6Ii8xODcwMDAxMDEvMzgyNDEyOTAwLTNhOTdmNDQ4LWM1ZmEtNDRjZS04NjUxLWZmNGM1YjJlYTc1ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyN1QwNjQ4MjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02OWNmMjEwOWNjN2FjYzViZTJjODg5NmJjMzA3NGYwYjQwYTRlZDI0YjcxNzgyYmJiYWFlZGYzYTJkNzk0ODI2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.hdu1iia2X4lWO7CyfdVHyCNxedhzX0tQOxK8dGdHO8M)
> 
> ### Actual results
> ![bambu lab screenshot](https://private-user-images.githubusercontent.com/187000101/382412945-4a894793-b165-49c2-8013-911bc0bb4718.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUyODI0MDgsIm5iZiI6MTczNTI4MjEwOCwicGF0aCI6Ii8xODcwMDAxMDEvMzgyNDEyOTQ1LTRhODk0NzkzLWIxNjUtNDljMi04MDEzLTkxMWJjMGJiNDcxOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyN1QwNjQ4MjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lN2I4NjExMWIxMDcwYzZkZmZmMjFhM2ZhMzcwZTdmYWE2ZjFiZmRmYjIzNWZiZTQ3YjUzNzZmYzQyMmQ3YTE0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.KyYr3iVipcxNkQNG1kH_8k-SZacTNkmQStk-ceaVx-g)
> 
> ### Expected results
> should not be black an able to see object and editing buttons
> 
> ### Project file & Debug log uploads
> its any file.Graphics
> 
> ### Checklist of files to include
> * [ ]  Log file
> * [ ]  Project file

hi, pls help to check if there is a context menu popped after right click in the 3d region, thanks.
like this:
[image]

Haidiye00, 2025-03-26:
https://github.com/bambulab/BambuStudio/issues/4419

ericLugtigheid, 2025-04-07:
[image]

I enabled FPS  and it worked again

DubDubFPV, 2026-09-26:
> amd software app... image sharpening for some reason seems to trigger this. I just turned everything off on the default profile and it reverted back to normal. odd because neither drivers or bambu had updated and everything was working fine after windows updated.

Even on later versions (such as the 02.08.02.60/1) its still an issue