bambulab/BambuStudio #5127: AMS status does not update correctly or slow to update on device screen

mytechguyri, 2024-11-02:
### Bambu Studio Version

Public Beta 1.10.0.81

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

AMD Ryzen 7 5700U

### Printer

Bambu Lab P1S

### How to reproduce

During a print with color changes, the AMS filament status doesn't always update, i.e. printer was printing black from AMS slot #2, changes filament to white from AMS Slot #1, but device screen continues to show that filament is black from AMS slot #2

### Actual results

printer is printing a different color than what is indicated on screen

### Expected results

device screen should update with he current tactive color, and filament path to the toolhead.

### Project file & Debug log uploads

This isn't related to a specific project

### Checklist of files to include

- [X] Log file
- [X] Project file

mytechguyri, 2024-11-02:
as shown in this attached photo... Printer is ACTUALLY currently printing black filament from AMS slot #4, and has been for several minutes, but the screen still shows the previous grey filament in AMS Slot #1 as the active filament.
[image]

walterwongbbl, 2024-11-12:
@mytechguyri 
Can you provide your studio log file.

mytechguyri, 2024-11-12:
Sorry, I went back to the release version

Locxion, 2024-11-22:
This also happens on the Stable Release Version: 1.10.1.50
[image]
[debug_Fri_Nov_22_16_25_57_31276.log](https://github.com/user-attachments/files/17873813/debug_Fri_Nov_22_16_25_57_31276.log)