bambulab/BambuStudio #5131: App crashes constantly on Mac when sending job

hackmiller, 2024-11-02:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Apple Macbook Pro Sequoia 15.1

### Additional system information

Apple M2 Max Chip
32 Gig memory

### Printer

X1 Carbon

### How to reproduce

Load a file -> Slice -> Send to printer. It will send the job and crash, almost every time, first time after sleep or starting for sure. Really annoying and have sent in a million crash reports

### Actual results

As mentioned, it crashed.

### Expected results

It should not crash...

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17608463/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Chompworks, 2025-03-04:
Also experiencing this. Seems to complete the transfer and then crash. I don't get it as consistently often as OP, perhaps 1 in 10 times after 'Print Plate' and sending the slice to one of my printers. Was present in last release and is still present in current version, 1.10.2.76.

this is on an M2 MacBook Air, running MacOS 11.7.10 (20G1427)

walterwongbbl, 2025-04-21:
Please use the latest version of Studio.

closed 2025-04-21 by walterwongbbl